commonsguy/cwac-cam2 #14: Support Tap-to-Focus

commonsguy, 2015-07-16:
In addition to automatic focus modes, support a tap on the screen background to trigger a manual focus. Offer an `IntentBuilder` method and corresponding extra to enable/disable this feature (initially enabled).

vxhviet, 2016-05-05:
Dear @commonsguy,

I would like to ask is the Tap to Focus feature is still available in version 0.5.9? There's only 3 focus modes in the demo Playground which are:
- Continuous.
- Off.
- Extended DOF.

Sorry if I'm missing something obvious but this is a really needed feature.
Thanks again for your time.

commonsguy, 2016-05-05:
> I would like to ask is the Tap to Focus feature is still available in version 0.5.9? 

This issue is still open.

crackTheCode, 2016-06-21:
Is there any hope to see this feature any time soon?

commonsguy, 2016-06-21:
@crackTheCode Well, it depends on what you are looking for.

This issue is for a fairly vanilla implementation, one where a tap anywhere on the preview would trigger a basic autofocus. That should be straightforward for the classic `Camera` engine, and I have not researched how it would be done with `android.hardware.camera2`, though presumably it can be done. I am not contemplating anything elaborate (e.g., focus on the point where the user tapped).

What exactly are you looking for?

crackTheCode, 2016-06-21:
I am new to the camera API so I don't know how to implement it the right way in both engines and I haven't found anything particularly helpful so I look forward to a tap to focus and exposure "calibration" feature in both engines, picture and video capture mode in a future release as the native camera app does on my OnePlus One Oxygen OS

commonsguy, 2016-06-21:
This library is for *very basic picture and video capture*, nothing more. This library will never have exposure calibration. This library will never match your native camera app feature for feature.