commonsguy/cwac-cam2 #15: Support EXIF Orientation Normalization

commonsguy, 2015-07-16:
If the image comes back with an EXIF orientation header, and if the client requests it (`IntentBuilder` method and corresponding extra), rotate the image and save the result sans orientation header. Do not bother implementing this until you can do an NDK version, as doing this in Java code sucks from a heap space standpoint.

commonsguy, 2015-07-20:
If the EXIF `Orientation` header is `1` (normal), secondary test is the actual orientation of the device. Have `OrientationPlugin` save this information for the capture request somewhere, so that the image cleanup process can pick it up and fix the image if needed. Test case is HTC M7 GPE per issue #45.

commonsguy, 2015-08-11:
Note that this needs to work even if auto-rotate mode is disabled (see issue #75)

leeeomaaax, 2016-05-10:
Hi there! Is this issue still going to be implemented?

commonsguy, 2016-05-10:
@leeeomaaax At some point, yes.

andrewprocter, 2016-05-10:
@leeeomaaax this may help you in the meantime

```
Bitmap resizedImage = resizeBitmap(getApplicationContext(), outputFileUri);

ExifInterface exif = new ExifInterface(outputFileUri.getPath());
int rotation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_NORMAL);
int rotationInDegrees = exifToDegrees(rotation);

Matrix matrix = new Matrix();
if (rotation != 0f) {matrix.preRotate(rotationInDegrees);}

if (rotationInDegrees != 0) {
// Rotate image
resizedImage = Bitmap.createBitmap(resizedImage, 0, 0, resizedImage.getWidth(), resizedImage.getHeight(), matrix, true);
}
```

```
private static int exifToDegrees(int exifOrientation) {
        if (exifOrientation == ExifInterface.ORIENTATION_ROTATE_90) { return 90; }
        else if (exifOrientation == ExifInterface.ORIENTATION_ROTATE_180) {  return 180; }
        else if (exifOrientation == ExifInterface.ORIENTATION_ROTATE_270) {  return 270; }
        return 0;
    }
```

commonsguy, 2016-05-10:
@andrewprocter Two downsides with doing this in Java are speed and heap limits. I used this approach in my original camera library, and that's why I want to use the NDK for this library.

Also, note that `outputFileUri.getPath()` only works for `file:` `Uri` values. The AOSP has some better EXIF code that can handle `InputStream`. Part of addressing this issue will be to incorporate that code, as I did in my original camera library.

andrewprocter, 2016-05-10:
Thanks @commonsguy! @leeeomaaax he's right on all counts. If you're in a pinch the code I posted will work, but please pay attention to the downsides @commonsguy mentioned.

danielgindi, 2016-05-16:
What is currently missing? EXIF reader in native code?

commonsguy, 2016-05-16:
@danielgindi What is currently missing is my devoting the time to the problem. This involves integrating some JNI code to do the actual bitmap rotation and integrating some EXIF reader code that does not rely upon a `File` (the way the built-in `ExifInterface` does). I am working on this for v0.6.0, due out in a couple of weeks.

danielgindi, 2016-05-16:
I have asked because I may be able to help.
I wrote these:
C# https://github.com/danielgindi/Imaging.net/blob/master/Imaging.net/Decoders/ImageDimensionsDecoder.cs / https://github.com/danielgindi/Imaging.net/blob/master/Imaging.net/ImagingUtility.cs#L158
ObjC https://github.com/danielgindi/DGFastImageSize/blob/master/DGFastImageSize.m

So the logic could be re-written (possibly by me) in native code for JNI.
There's both the mapping if the EXIF tag to the correct rotation to perform, and there's a native reader that traverses the EXIF table, finds the correct tag, and rotates the image size to receive correct image size (used for other cases, but the tag itself is important)

commonsguy, 2016-05-16:
@danielgindi: If you wanted to publish some Android library that offers EXIF normalization, I might link to it.

The use case is: given a `byte[]` of JPEG data, end up with a `byte[]` of JPEG data with the properly-rotated image and ideally all the original EXIF tags (excluding the orientation tag and possibly the thumbnail). Bonus points if the caller can provide a hint as to what to use for the orientation as an override, for devices like the HTC M7 GPE that [appear to improperly encode the JPEG](https://github.com/commonsguy/cwac-cam2/issues/45).

In terms of direct contributions to this library, EXIF normalization would be [too large](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-code).

My current plan is to use [this library](https://github.com/sephiroth74/Android-Exif-Extended) (culled from AOSP sources) for EXIF reading/writing and a cut-down version of [this code](https://github.com/AndroidDeveloperLB/AndroidJniBitmapOperations) for the rotation, assuming that I determine that it is faster or less memory-intensive than is using a `Matrix` in Java. I have this working in a proof-of-concept mode in a standalone app for an upcoming version of [my book](https://commonsware.com/Android).

Thanks for your interest!

danielgindi, 2016-05-16:
Well in that case it seems that you are all set :)

I think that rotation should be done in native code anyway, as in Java you don't get any memory allocated enough to work on any bitmap.

danielgindi, 2016-05-16:
I'm aware of the HTC problem. Ate sh*t because of that. But it's not a single model, it's most HTC modes. Their ROMs are faulty.

danielgindi, 2016-05-16:
Btw, according to http://stackoverflow.com/questions/543402/can-a-jpeg-compressed-image-be-rotated-without-a-loss-in-quality, we may be able to rotate the JPEG without loosing quality

commonsguy, 2016-05-16:
@danielgindi That's where some native JPEG rotation code would come in handy. The code that I am looking to use rotates a `Bitmap`, not a `byte[]` of JPEG data. So, I wind up with:

- the original JPEG `byte[]`
- the decoded `Bitmap`
- the rotated `Bitmap`

at minimum, all in the Java heap. This is no worse than using a `Matrix` to rotate the `Bitmap` and possibly is better. But I am not an image encoding/decoding expert, and I am not aware of any Android-friendly JPEG-manipulation libraries, such as ones that could do the JPEG rotations per your most recent comment. So, I work with the libraries that are at my disposal.

To get back to my earlier comment, if you wanted to write a library that could do this work faster/better/with less memory, I'd be happy to try it. But that really should be an independent library, as that work is not somehow unique to CWAC-Cam2.

danielgindi, 2016-05-16:
There's already a library that does that, apparantly:
http://stackoverflow.com/questions/706665/lossless-jpeg-rotate-90-180-270-degrees-in-java

But bear in mind that not all rotations can be performed this way, and not all images could be rotated this way. So there needs to be a fallback

commonsguy, 2016-05-16:
@danielgindi By my read, that code has not been touched in a decade. It pre-dates Android. [This library](https://github.com/bkhall/AndroidMediaUtil) based on MediaUtil might work.

danielgindi, 2016-05-17:
Except it's not JNI. But it is possible to use the same algorithms and blindly converting them to native code

commonsguy, 2016-05-17:
Well, in a performance test between [this JNI code](https://github.com/AndroidDeveloperLB/AndroidJniBitmapOperations) and using `Matrix`, the JNI is only ~5% faster in tests on Android 4.4 and Android 6.0. With that in mind, I will stick with `Matrix` for now.

commonsguy, 2016-05-29:
Addressed in v0.6.0.

closed 2016-05-29 by commonsguy

shahzadyounas, 2016-07-19:
Hi

I am on the latest version 0.6.4 - however still seeing this issue on Galaxy S4 Mini and Lenovo TAB S8-50F .  My app requires use of the front facing camera only in portrait mode:

 Intent i=new CameraActivity.IntentBuilder(RegistrationStage_5.this)
                        .facing(Facing.FRONT)
                        .facingExactMatch()
                        .to(new File(getExternalCacheDir(), "portrait-front.jpg"))
                        .orientationLockMode(OrientationLockMode.PORTRAIT)
                        .zoomStyle(ZoomStyle.NONE)
                        .updateMediaStore()
                        .build();

                startActivityForResult(i,  REQUEST_PORTRAIT_FFC);

However in emulator, and samsung S4 mini and Lenovo Tablet all previews are rotated 90degrees?

commonsguy, 2016-07-19:
@shahzadyounas This would be a separate issue, as this issue has nothing to do with previews. It has to do with the actual images being saved.

I suggest that you try commenting out `orientationLockMode(OrientationLockMode.PORTRAIT)` and see what your behavior is. Then, open a fresh issue with your symptoms. Also include the `Build.PRODUCT` values for both devices &mdash; particularly for the Samsung, probably there are several models that they marked as "S4 Mini".

shahzadyounas, 2016-07-19:
Yes I tried with orientation lock removed as well as mirrorpreview removed and present.
All yielded the same issue.  

I've seen various posts mentioning how FFC does not save EXIF data on some models:

http://stackoverflow.com/questions/13245556/exif-orientation-tag-value-always-0-for-image-taken-with-portrait-camera-app-and

But I wonder why even in the v23 emulator I get the same issue (preview image is rotated).

If we can get it to work in the emulator for this issue I am sure it will work for the phone models causing issues.

commonsguy, 2016-07-19:
> If we can get it to work in the emulator for this issue I am sure it will work for the phone models causing issues

I never test on the emulator. Users do not use the emulator, and the closer the code is to hardware, the less faithful the emulator becomes.

Please open a fresh issue with your symptoms. Also include the Build.PRODUCT values for both devices — particularly for the Samsung, probably there are several models that they marked as "S4 Mini".

shahzadyounas, 2016-07-19:
Thanks - issue created.