commonsguy/cwac-cam2 #21: Theming Support and Instructions

commonsguy, 2015-07-16:
Things like the FAB color ought to be configurable by themes. Implement this and write up instructions for how to theme the `CameraActivity` and its contents.

commonsguy, 2015-07-20:
Note that `CameraActivity` needs an action bar (for `ConfirmationFragment`) and so cannot work with themes that outright eliminate the action bar. This includes "toolbar-based themes" (see issue #48).

yazeed44, 2015-07-20:
Why not work with toolbar instead of actionbar ?

commonsguy, 2015-07-20:
@yazeed44 That makes the problem worse, not better. I would then have to have three versions of the library: one for native Android 5.0 `Toolbar`, one for the `Toolbar` backport in `appcompat-v7` but still using native activities/fragments, and one for the `Toolbar` backport in `appcompat-v7` where I use `AppCompatActivity` and fragment backports.

yazeed44, 2015-07-21:
I don't understand ? why not use the support libraries all the time ?
It shouldn't be much of a difference than the native

commonsguy, 2015-07-21:
Sorry, I had a mistake in my previous comment, as AFAIK, the `Toolbar` backport requires `AppCompatActivity` and `Theme.AppCompat`. That means that using `Toolbar` is not possible, as I cannot cover all scenarios. I can use the native `Toolbar` with Android 5.0+ activities, and I can use the backported toolbar with `AppCompatActivity`, but I have no solution for Android 4.x activities (other than via `AppCompatActivity`).

> why not use the support libraries all the time ?

That does not meet my own personal requirements. And since I am the guy writing the code, my own objectives and requirements are fairly important to me.

yazeed44, 2015-07-21:
So in gist , will we be able to work with this lib with toolbar soon ?

commonsguy, 2015-07-21:
That depends on what you mean by "with toolbar". Issue #49 is all I am doing in this area "soon", for short values of "soon".