commonsguy/cwac-cam2 #38: Dependency dosen't fetch

yazeed44, 2015-07-17:
repositories {
    mavenCentral()
    maven {
        url "https://repo.commonsware.com.s3.amazonaws.com"
    }

}

dependencies {
    compile 'com.commonsware.cwac:cam2:0.1.+'
}

commonsguy, 2015-07-17:
I do not know what "dependency doesn't fetch" means. In the future, please provide a more meaningful description of your symptoms.

If by "dependency doesn't fetch", you mean "Gradle says that it cannot download the dependency", I cannot reproduce your problem:

```groovy
apply plugin: 'com.android.application'

android {
    compileSdkVersion 22
    buildToolsVersion "23.0.0 rc2"

    defaultConfig {
        applicationId "com.commonsware.myapplication"
        minSdkVersion 22
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}

repositories {
    maven {
        url "https://repo.commonsware.com.s3.amazonaws.com"
    }
}

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    compile 'com.commonsware.cwac:cam2:0.1.+'
}
```

You can reproduce my test by creating a new project in the Android Studio new-project wizard and modifying the `app/build.gradle` file to match the above file.

For an existing project, you may need to [refresh your dependency cache](https://docs.gradle.org/current/userguide/dependency_management.html#cache_refresh).

If you are still encountering problems after all of the above, please provide a more detailed description of your symptoms.

yazeed44, 2015-07-17:
I used ```./gradlew --refresh-dependencies```

This is the output

 Could not find any version that matches com.commonsware.cwac:cam2:0.1.+.

commonsguy, 2015-07-17:
@yazeed44 Please post the entire Gradle output.

yazeed44, 2015-07-17:
```gradle
./gradlew --refresh-dependencies
Parallel execution with configuration on demand is an incubating feature.
                                                                                                                                                                                                                        
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugCompile'.
   > Could not find any version that matches com.commonsware.cwac:cam2:0.1.+.
     Required by:
         MultiImagePicker:app:unspecified > net.yazeed44.imagepicker:imagepicker:1.1.1

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 1 mins 3.926 secs

```

commonsguy, 2015-07-17:
That's odd. Usually, it tells you where it is searching, via a list of URLs, and that is what I was looking to check.

You can try dropping back to `http` instead of `https` and see if you are running into some SSL problem.

You might also try visiting some of the relevant URLs in a Web browser, to confirm that you can access them:

http://repo.commonsware.com.s3.amazonaws.com/com/commonsware/cwac/cam2/maven-metadata.xml
http://repo.commonsware.com.s3.amazonaws.com/com/commonsware/cwac/cam2/0.1.0/cam2-0.1.0.pom

yazeed44, 2015-07-17:
I tried using http and the result is the same

https://github.com/yazeed44/MultiImagePicker/blob/e0a5b39f308f17048b2f4fbbb72d8716bfc0b047/imagepicker/build.gradle

this is whole source code

Also the two links works perfectly

commonsguy, 2015-07-17:
You can try replacing `0.1.+` with `0.1.0`, though that should not make a difference.

You can try downloading or cloning the repo and building one of the demo projects, modifying its `build.gradle` file to use `compile` instead of `releaseCompile` for the `com.commonsware.cwac:cam2:0.1.+` dependency (and removing the corresponding `debugCompile`), and see what you get.

Beyond that, I am starting to run out of ideas as to what may be going on here.

yazeed44, 2015-07-17:
When i used 0.1.0 
```gradle
What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugCompile'.
   > Could not find com.commonsware.cwac:cam2:0.1.0.
     Searched in the following locations:
         file:/home/yazeed44/Desktop/downloadLinux/android-studio/android-sdk-linux/extras/android/m2repository/com/commonsware/cwac/cam2/0.1.0/cam2-0.1.0.pom
         file:/home/yazeed44/Desktop/downloadLinux/android-studio/android-sdk-linux/extras/android/m2repository/com/commonsware/cwac/cam2/0.1.0/cam2-0.1.0.jar
         file:/home/yazeed44/Desktop/downloadLinux/android-studio/android-sdk-linux/extras/google/m2repository/com/commonsware/cwac/cam2/0.1.0/cam2-0.1.0.pom
         file:/home/yazeed44/Desktop/downloadLinux/android-studio/android-sdk-linux/extras/google/m2repository/com/commonsware/cwac/cam2/0.1.0/cam2-0.1.0.jar
     Required by:
         MultiImagePicker:app:unspecified > net.yazeed44.imagepicker:imagepicker:1.1.1

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

```

yazeed44, 2015-07-17:
Why not upload it on jcenter ?
jCenter works perfectly with me

commonsguy, 2015-07-17:
For some reason, it is not picking up *any* of your repositories. It's not even looking in `mavenCentral()`. It is only looking in those from the Gradle for Android plugin.

You might try moving your `repositories` and `dependencies` closures before the `android` plugin. You might also try creating a scrap project in Android Studio, putting in the dependency there, and confirming that it works.

yazeed44, 2015-07-18:
I tried it on another project and it works fine

commonsguy, 2015-07-18:
Then the problem is unique to your original project. Something about your complex dependency/repository setup in that project is causing you difficulty.

I am going to close this issue, as this is not demonstrably a problem with this library or its entry in the CWAC artifact repository. If you find clear evidence that it *is* a problem with this library or the repository, feel free to add a new comment, and I can reopen the issue.

closed 2015-07-18 by commonsguy

yazeed44, 2015-08-22:
It seems it's because of ```apply plugin: 'com.android.library' ```

I might have to make users add cwac-cam2 dependency manually

danhawkes, 2016-03-03:
Possibly related: http://stackoverflow.com/questions/30781104/gradle-plugin-com-android-library-does-not-search-in-custom-maven-repositories

Only workaround I've found is to add it to `allProjects` in the root `build.gradle`:

```
allprojects {
    repositories {
        jcenter()
        maven {
          url "https://repo.commonsware.com.s3.amazonaws.com"
        }
    }
}
```

jerzykiler, 2016-12-15:
The maven url should be added outside of buildscript like so:

buildscript {
    repositories {
        maven { url 'https://maven.fabric.io/public' }
    }

    dependencies {
        classpath 'io.fabric.tools:gradle:1.+'
    }
}
apply plugin: 'com.android.application'
apply plugin: 'io.fabric'

repositories {
    maven { url 'https://maven.fabric.io/public' }
    maven { url "https://s3.amazonaws.com/repo.commonsware.com" }
}