commonsguy/cwac-cam2 #39: NullPointerException before camera release

ppamorim, 2015-07-17:
This happens only when I click at the "take picture" button until the camera isn't released.

    java.lang.NullPointerException
            at      com.commonsware.cwac.cam2.ClassicCameraEngine$2.run(ClassicCameraEngine.java:134)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
            at java.lang.Thread.run(Thread.java:841)

ppamorim, 2015-07-17:
add syncronize(object) and notify can solve this issue?

commonsguy, 2015-07-17:
I doubt it. `session` was `null` when I called `takePicture()`, for some reason.

Could you please explain, in greater detail, how you are causing this crash? I do not know what "I click at the "take picture" button until the camera isn't released" means.

commonsguy, 2015-07-18:
Well, I should be checking `session` for `null` before I call `takePicture()` on the `CameraEngine` anyway, so I have added that check. This does not address why `takePicture()` would have wound up being called in this state in the first place, though I am guessing that issue #44 is related.

commonsguy, 2015-07-26:
Released in v0.1.1.

closed 2015-07-26 by commonsguy