commonsguy/cwac-cam2 #42: Changing CameraFragment Screws Up Preview Aspect Ratio

commonsguy, 2015-07-18:
HTC M7 GPE, switching from RFC to FFC in portrait results in stretched preview.

Nexus 4, rotating from portrait to landscape while showing RFC results in stretched preview.

commonsguy, 2015-07-25:
FYI, camera2 preview stretching comments from fadden: http://stackoverflow.com/questions/31614792/surfacetexture-preview-for-camera2-stretches-on-some-devices

tplkn, 2015-08-05:
This commit add zoom for camera view in landscape mode (for the same distance for pen)
- before land: http://take.ms/Pey6c (correct)
- after port: http://take.ms/UjtD4 (correct) land: http://take.ms/9HNSb (wrong)

nexus 5, android 5.1

commonsguy, 2015-08-05:
@tplkn AFAIK, that will be a natural side-effect of the full-screen preview. If you have suggestions on how to improve the algorithm, I am certainly up for any ideas. Note that I just committed another change, to take into account default-landscape devices, per issue #64. Thanks!

commonsguy, 2015-08-07:
Released in v0.2.0

closed 2015-08-07 by commonsguy