commonsguy/cwac-cam2 #44: Disable FAB While Picture-Taking In Progress

commonsguy, 2015-07-18:
Once the user taps the FAB, it (and the FAM) should be disabled, to prevent attempts to change things while the picture-taking is in progress. Only if we return to the fragment (e.g., user requests a re-take in the `ConfirmationFragment`) should we re-enable these.

ppamorim, 2015-07-20:
What do you think to use syncronize(object) to fix this issue?

commonsguy, 2015-07-20:
This is not a threading problem. Even to the extent that we try to treat it as a threading problem, one of the threads in question is the main application thread, and we do **not** want to be blocking the main application thread waiting for photo processing.

ppamorim, 2015-07-20:
This is a little bit similar of #39, but I tested to sync that param and works fine at #39.

commonsguy, 2015-07-26:
Released in v0.1.1.

closed 2015-07-26 by commonsguy