commonsguy/cwac-cam2 #50: Own CameraFragment support

gbhall, 2015-07-21:
Hi there, 

I love your work. I've been following your camera support closely and wanted to try this new library as I had some difficulties with the old cwac-camera where some newer devices were crashing, e.g. HTC One M9 with `RuntimeException: setParameters failed`.

Even though cwac-cam2 seems to be a different approach, I'm trying to replicate the ability to create my own `CameraFragment` class like before, so that in my activity I can have my own controls, and display the camera fragment when I want to. I then call fragment.takepicture() from the activity when I want to take a picture. Before with cwac-camera I was overriding `getPhotoPath()` for `CameraFragment` so I knew where to retrieve the images to do with what I wanted later on.

 At the moment, for `cwac-cam2` I'm having an issue where I think I need to override `prepController()` in `CameraFragment` however I can't because it's private.

This is the error I get:
```
07-21 23:04:31.721  28070-28070/com.gbhall.reef E/Event? Could not dispatch event: class com.commonsware.cwac.cam2.CameraController$ControllerReadyEvent to subscribing class class com.gbhall.reef.ReefCamera.ReefCameraFragment
    java.lang.NullPointerException: Attempt to invoke virtual method 'android.view.View android.view.ViewGroup.getChildAt(int)' on a null object reference
            at com.commonsware.cwac.cam2.CameraFragment.prepController(CameraFragment.java:220)
            at com.commonsware.cwac.cam2.CameraFragment.onEventMainThread(CameraFragment.java:208)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at de.greenrobot.event.EventBus.invokeSubscriber(EventBus.java:498)
            at de.greenrobot.event.EventBus.postToSubscription(EventBus.java:433)
            at de.greenrobot.event.EventBus.postSingleEventForEventType(EventBus.java:410)
            at de.greenrobot.event.EventBus.postSingleEvent(EventBus.java:383)
            at de.greenrobot.event.EventBus.post(EventBus.java:263)
            at com.commonsware.cwac.cam2.CameraController.onEventMainThread(CameraController.java:261)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at de.greenrobot.event.EventBus.invokeSubscriber(EventBus.java:498)
            at de.greenrobot.event.EventBus.invokeSubscriber(EventBus.java:492)
            at de.greenrobot.event.HandlerPoster.handleMessage(HandlerPoster.java:67)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5254)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
```

This is what I've tried to do so far:
```
package com.gbhall.reef.ReefCamera;

import android.app.Activity;
import android.content.ClipData;
import android.content.Context;
import android.net.Uri;
import android.os.Build;
import android.os.Bundle;
import android.app.Fragment;
import android.os.Environment;
import android.provider.MediaStore;
import android.support.design.widget.Snackbar;
import android.view.LayoutInflater;
import android.view.View;
import android.view.ViewGroup;

import com.commonsware.cwac.cam2.CameraController;
import com.commonsware.cwac.cam2.CameraEngine;
import com.commonsware.cwac.cam2.CameraFragment;
import com.commonsware.cwac.cam2.CameraSelectionCriteria;
import com.commonsware.cwac.cam2.CameraView;
import com.commonsware.cwac.cam2.PictureTransaction;
import com.gbhall.reef.R;

import java.io.File;
import java.text.SimpleDateFormat;
import java.util.Date;
import java.util.LinkedList;

public class ReefCameraFragment extends CameraFragment {

    private CameraController mController;
    private ViewGroup mPreviewStack;

    public ReefCameraFragment() {
    }

    public void takePicture() {
        PictureTransaction transaction = new PictureTransaction.Builder().build();
        mController.takePicture(transaction);
    }

    @Override
    public void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        // Create a new camera controller and set it the fragment
        mController = new CameraController();
        CameraSelectionCriteria criteria = new CameraSelectionCriteria.Builder().facing(CameraSelectionCriteria.Facing.BACK).build();
        mController.setEngine(CameraEngine.buildInstance(getActivity()), criteria);
        setController(mController);
    }

    @Override
    public View onCreateView(LayoutInflater inflater, ViewGroup container,
                             Bundle savedInstanceState) {
        View rootView = inflater.inflate(R.layout.element_reefcamera_layout, container, false);
        mPreviewStack = (ViewGroup) rootView.findViewById(R.id.cwac_cam2_preview_stack);

        // Add preview to fragment
        if (mController != null && mController.getNumberOfCameras() > 0) {
            prepController();
        }

        return rootView;
    }

    private void prepController() {
        LinkedList<CameraView> cameraViews = new LinkedList<>();
        CameraView cv = (CameraView) mPreviewStack.getChildAt(0);

        cameraViews.add(cv);

        for (int i=1; i < mController.getNumberOfCameras(); i++) {
            cv=new CameraView(getActivity());
            cv.setVisibility(View.INVISIBLE);
            mPreviewStack.addView(cv);
            cameraViews.add(cv);
        }

        mController.setCameraViews(cameraViews);
    }

}
```

Cheers, and I hope this is the right place to post this. Sorry if it's not!

Thanks for reading :)

commonsguy, 2015-07-21:
> I love your work.

Um, thanks!

> I'm trying to replicate the ability to create my own CameraFragment class like before

As is noted on the project `README`, the only documented and supported API is by means of starting the `CameraActivity`. Everything else inside of there is in a serious state of flux, and will be for the foreseeable future. I desperately want to constrain the "surface area" of the API to something that I am in position to support well, as simply getting the implementation to be solid is going to be enough of a challenge.

> I'm having an issue where I think I need to override prepController() in CameraFragment however I can't because it's private

Make a fork of the library, modify `prepController()` to be `protected` in your fork, and see if it allows you to solve your problem. If it does, let me know, and I can make it `protected` in the official library.

I don't have an issue in general with trivial changes like that, but I do want to establish a pattern where I only make such changes when we both know that it solves a problem, not where it is merely a "maybe".

But I must caution you again: do not assume that the `CameraFragment` API is anywhere near stable. I guarantee you that there will be changes, possibly substantial ones, in time.

gbhall, 2015-07-21:
Thank you. Duly noted regarding stability. I'll try later today and report back if that works. 

Cheers.

gbhall, 2015-07-22:
I haven't had a chance yet, but I promise I will get back to you. 

Cheers.

gbhall, 2015-07-28:
Problem solved by overriding ``onEventMainThread(CameraController.ControllerReadyEvent event)`` for ``CameraFragment``:

```java
    @Override
    public void onEventMainThread(CameraController.ControllerReadyEvent event) {
        if(event.isEventForController(this.mController)) {
            this.prepController();
        }
    }
```

closed 2015-07-28 by commonsguy