commonsguy/cwac-cam2 #57: Add Video Support for CameraTwoEngine

commonsguy, 2015-07-30:
Existing stuff is only for `ClassicCameraEngine`. Need an implementation for `CameraTwoEngine`, for Android 5.0+ support.

commonsguy, 2015-09-18:
Notes on preview for camera2 for video recording: http://stackoverflow.com/questions/32611152/how-to-setup-camera-preview-with-camera2-api-for-both-photo-and-video-at-once/32659744#32659744

bpappin, 2015-10-16:
Any ETA on this? 
The method CameraTwoEngine.stopVideoRecording(CameraSession) is empty, so the stop doesn't actually work.

Note: this is also likely the cause of #82

commonsguy, 2015-10-16:
@bpappin It's supposed to be wired such that at the moment, force-classic mode is being used for all video, such that `CameraTwoEngine` would not be used. It's possible I lost that logic somewhere, as I'm not seeing it at the moment, but I'll look at it more closely tomorrow.

commonsguy, 2015-10-17:
@bpappin That indeed appears to be the case, where I lost my code to force video recording to use the classic engine. This enhancement, to implement camera2-based video recording, is still outstanding, but I will be patching issue #82 shortly by forcing the classic engine.