commonsguy/cwac-cam2 #62: java.lang.IllegalStateException: CameraDevice was already closed

gbhall, 2015-08-01:
This is all the info I have from an error report. Note the app was minified with ProGuard.

```
java.lang.IllegalStateException: CameraDevice was already closed
at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:1482)
at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:425)
at com.commonsware.cwac.cam2.aj.onOpened(Unknown Source)
at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:118)
at android.os.Handler.handleCallback(Handler.java:739)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:135)
at android.os.HandlerThread.run(HandlerThread.java:61)
```

commonsguy, 2015-08-01:
Sorry, but there is nothing I can really do with that. Occasionally, just a stack trace will be sufficient for me to deduce the nature of a problem. Most of the time, I need to know the exact device plus the steps to reproduce the problem.

BTW, I would not consider a 0.1.x version of a library to be something worthy of using in production.

JulianCardona, 2016-01-25:
i just Got this exact problem, it can be reproduced by pressing back from the camera activity and going in again quickly.

commonsguy, 2016-01-26:
@JulianCardona Please provide the rest of the information requested in [the contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/CONTRIBUTING.md). In particular, I need to know what app you are using that reproduces the problem and I need to know what device you are using that reproduces the problem. Thanks!

msisinni, 2016-04-13:
I was able to reproduce this somewhat inconsistently in my app. I say inconsistently because there is also another non-fatal bug produced using the same steps that instead results in the camera being in use and further requests to open the camera failing (silently - the library just closes the new Activity without any message). Trying to open another camera app after that non-fatal bug (without first clearing my app) fails too, since the camera is already in use.
I am testing with an HTC One M7 running 5.0.2, and I have been able to reproduce this using the library versions 0.5.3 and 0.5.6. I also have Crashlytics reports from two users: one with a Samsung S5, and the other with a Samsung Edge S6.

The trick to reproducing it is to spam the button used to launch the camera while the app is already skipping frames. Opening the camera, backing out, and then rapidly clicking the button seems like the easiest way to do this.

Here is the code used to launch the camera. It is tied to a View OnClickListener:

        Intent cameraIntent = new CameraActivity.IntentBuilder(getActivity())
            .skipConfirm()
            .to(new File(getUnprocessedImageFilePath()))
            .build();

        startActivityForResult(cameraIntent, AppConstants.TAKE_PICTURE_REQUEST_CODE);

My stack trace is almost the same as above: 

    Fatal Exception: java.lang.IllegalStateException: CameraDevice was already closed
       at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:1482)
       at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession(CameraDeviceImpl.java:425)
       at com.commonsware.cwac.cam2.CameraTwoEngine$InitPreviewTransaction.onOpened(CameraTwoEngine.java:430)
       at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:118)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:155)

commonsguy, 2016-04-13:
@msisinni 

> spam the button used to launch the camera

To confirm, "spam the button" means to press it repeatedly in rapid succession? And "the button" in this case is in your UI, where you make the `startActivityForResult()` call?

If so, have you been able to reproduce this with the `demo-playground` app? You should be able to spam the action bar item in much the same way.

> I am testing with an HTC One M7

What is the `Build.PRODUCT` value? There are a few variants of the HTC One M7. I have one (`m7_google`), but it would help to know which one you have, particularly since my M7 is running 5.1.

msisinni, 2016-04-13:
Yes, rapidly press the bit of UI whose OnClickListener creates that Intent and calls startActivityForResult().
No, I have not tested this with the demo app.
My phone's Build.PRODUCT returns "cingular_us"

commonsguy, 2016-04-13:
@msisinni I'll see if I can reproduce it on my M7 and other devices akin to the ones you cited from your Crashlytics reports. In the interim, you might consider disabling that button after clicking it, enabling it in `onActivityResult()`.

msisinni, 2016-04-14:
I did actually try disabling it. The first line in my OnClickListener disables the button. During the large amount of lag after opening and closing the camera several times, it looks like I can click the button multiple times before setting off one of the two bugs (multiple ripple effects). The code in the OnClickListener is only set off once, though - I added log statements for the onClick and onActivityResult, and I'm not getting log statements twice in a row anywhere.

commonsguy, 2016-04-14:
I cannot reproduce the problem, on my M7 or on an S5, by spamming the `demo-playground` action bar item, using a configuration matching your `IntentBuilder` code. I will leave this issue open for a while, as I may try to write a separate app that more closely mirrors yours.

VBegoyan, 2017-02-28:
guys, anyone has already the answer ?

VBegoyan, 2017-02-28:
I got the same exception sometimes but very often, and can't find the solution. 
Any ideas?

commonsguy, 2017-02-28:
@VBegoyan Please follow [the contribution guidelines for bug reports](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports). Provide a way for me to reproduce the problem. **Do not** simply chime in with "me too!" comments.

asata91, 2017-08-31:
I have the same problem, When i am trying to switch from portrait mode to landscape, it throws that error

commonsguy, 2017-08-31:
@asata91 Please follow [the contribution guidelines for bug reports](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports). Provide a way for me to reproduce the problem. **Do not** simply chime in with "me too!" comments.

Also, since this library will be discontinued within a few weeks, I suggest you move to something else, such as CameraKit-Android or Fotoapparat.

asata91, 2017-09-01:
Hi all, i have implemented all as described in documentation  and my flow is like that: 
All works great, but when i am capturing picture in portrait mode , than flipping devices to landscape and capturing also in landscape mode it crashes. 
Portrait -> Landscape  = Crash
Portrait -> Landscape -> Portrait = Works Excellent !

Any ideas
p.s sorry for my english

asata91, 2017-09-01:
Easiest way to reproduce it is that , i quickly changing from portrait mode to landscape and it crash and throws that exception 

`
09-01 23:14:26.599 5722-7232/com.essentialsln.memtalk E/AndroidRuntime: FATAL EXCEPTION: CameraBackground
                                                                        Process: com.essentialsln.memtalk, PID: 5722
                                                                        java.lang.IllegalStateException: CameraDevice was already closed
                                                                            at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:1984)
                                                                            at android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:825)
                                                                            at android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:880)
                                                                            at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:231)
                                                                            at com.essentialsln.memtalk.ui.fragments.Camera2Fragment$11.onConfigured(Camera2Fragment.java:797)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
                                                                            at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
                                                                            at android.os.Handler.handleCallback(Handler.java:739)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                            at android.os.Looper.loop(Looper.java:148)
                                                                            at android.os.HandlerThread.run(HandlerThread.java:61)`

commonsguy, 2017-09-01:
@asata91 First, it does not crash with either the `demo/` or `demo-playground/` apps. Second, my library does not have a `com.essentialsln.memtalk.ui.fragments.Camera2Fragment` class in it.

closed 2017-09-01 by commonsguy