commonsguy/cwac-cam2 #68: Runtime exception when call Camera#open

tplkn, 2015-08-07:
In come cases application cannot to camera (event official camera app)

Error message:
Fatal Exception: java.lang.RuntimeException
Fail to connect to camera service

https://github.com/commonsguy/cwac-cam2/blob/master/cam2/src/main/java/com/commonsware/cwac/cam2/ClassicCameraEngine.java#L67

Expected result: avoid application crash

commonsguy, 2015-08-07:
Please indicate:

- the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem
- the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish)
- the steps necessary, using that code and that hardware, to reproduce the problem

tplkn, 2015-08-07:
Nexus 5, android 5.1.1
cannot provide other information at the mometn

commonsguy, 2015-09-05:
Closing as not reproducible. I need you to give me the code necessary to reproduce the problem (this could be one of the demo apps, or it could be some demo project of yours that you publish) and
the steps necessary, using that code and that hardware, to reproduce the problem. If you have that information, add it in a comment, and I can reopen the issue.

abkaplan07, 2016-09-22:
Request to reopen this issue. We've observed this exact issue from a user on a Samsung Galaxy S4 (GT-I9505) running Lollipop (Android 5.0.1). Errors came in for both Video and Camera intents:

```
Intent cameraIntent = new CameraActivity.IntentBuilder(mActivity)
        .to(file).skipConfirm().build();
```
and
```
Intent videoIntent = new VideoRecorderActivity.IntentBuilder(mActivity)
        .to(file).durationLimit(secs * 1000).build()
```

Our team used to use `Camera.open()` to inspect the device and verify photos/video returned from the normal Android media intents, and often observed this same error in the field from Bugsnag reporting. On some devices we could reproduce the error, but not on a consistent basis. I suspect that hardware/resource locking may be the root cause of this issue.

IMO this error should be caught and a message returned in the activity result.

commonsguy, 2016-09-22:
@abkaplan07 Can you at least give me a stack trace?

abkaplan07, 2016-09-22:
Sure:
```
java.lang.RuntimeException Fail to connect to camera service 
    Camera.java:545 android.hardware.Camera.<init>
    Camera.java:385 android.hardware.Camera.open
    ClassicCameraEngine.java:78 com.commonsware.cwac.cam2.ClassicCameraEngine$1.run
    ThreadPoolExecutor.java:1112 java.util.concurrent.ThreadPoolExecutor.runWorker
    ThreadPoolExecutor.java:587 java.util.concurrent.ThreadPoolExecutor$Worker.run
    Thread.java:818 java.lang.Thread.run
```

commonsguy, 2016-09-26:
Addressed in v0.6.8.

closed 2016-09-26 by commonsguy