commonsguy/cwac-cam2 #71: Buffers exceeded

gbhall, 2015-08-09:
Pressed 'Capture' in CWAC-Cam2 Demo (release version 0.2.0) on device: HTC Sensation; Android 4.0.3

Trace:
```
08-10 00:41:52.591  29739-30254/com.commonsware.cwac.cam2.demo D/memalloc﹕ virtual int gralloc::PmemKernelAlloc::map_buffer(void**, size_t, int, int): Mapped 3133440 bytes
08-10 00:42:00.659  29739-30254/com.commonsware.cwac.cam2.demo E/SurfaceTexture﹕ [unnamed-29739-0] setBufferCount: client owns some buffers
08-10 00:42:00.659  29739-30254/com.commonsware.cwac.cam2.demo E/SurfaceTexture﹕ [unnamed-29739-0] dequeueBuffer: MIN_UNDEQUEUED_BUFFERS=2 exceeded (dequeued=5)
08-10 00:42:00.669  29739-30254/com.commonsware.cwac.cam2.demo W/dalvikvm﹕ threadid=11: thread exiting with uncaught exception (group=0x40ab1228)
08-10 00:42:00.679  29739-30254/com.commonsware.cwac.cam2.demo E/AndroidRuntime﹕ FATAL EXCEPTION: pool-2-thread-1
    java.lang.RuntimeException: startPreview failed
            at android.hardware.Camera.startPreview(Native Method)
            at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:302)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1076)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:569)
            at java.lang.Thread.run(Thread.java:864)
08-10 00:42:00.759  29739-29739/com.commonsware.cwac.cam2.demo D/memalloc﹕ /dev/pmem: Mapped buffer base:0x5bae3000 size:18452480 offset:16363520 fd:97
08-10 00:42:00.789  29739-29739/com.commonsware.cwac.cam2.demo D/memalloc﹕ /dev/pmem: Mapped buffer base:0x5ccef000 size:20541440 offset:18452480 fd:100
08-10 00:42:00.919  29739-29741/com.commonsware.cwac.cam2.demo D/dalvikvm﹕ GC_CONCURRENT freed 1305K, 31% free 3275K/4707K, paused 2ms+5ms
08-10 00:42:00.959  29739-29739/com.commonsware.cwac.cam2.demo D/memalloc﹕ /dev/pmem: Mapped buffer base:0x5e086000 size:28897280 offset:26808320 fd:63
08-10 00:42:00.999  29739-29739/com.commonsware.cwac.cam2.demo D/SensorManager﹕ [SensorManager] registerListener: delay = 200000
08-10 00:42:01.009  29739-30496/com.commonsware.cwac.cam2.demo W/dalvikvm﹕ threadid=13: thread exiting with uncaught exception (group=0x40ab1228)
08-10 00:42:01.009  29739-30496/com.commonsware.cwac.cam2.demo E/AndroidRuntime_2_crash﹕ crash in the same process: pool-2-thread-2
    java.lang.RuntimeException: Fail to connect to camera service
            at android.hardware.Camera.native_setup(Native Method)
            at android.hardware.Camera.<init>(Camera.java:307)
            at android.hardware.Camera.open(Camera.java:268)
            at com.commonsware.cwac.cam2.ClassicCameraEngine$3.run(ClassicCameraEngine.java:176)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1076)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:569)
            at java.lang.Thread.run(Thread.java:864)
08-10 00:42:01.029  29739-29739/com.commonsware.cwac.cam2.demo D/memalloc﹕ /dev/pmem: Mapped buffer base:0x5fc15000 size:30986240 offset:28897280 fd:108
```

commonsguy, 2015-08-09:
@gbhall There are a few device models that are described as an "HTC Sensation". Can you tell me what the `Build.PRODUCT` value is?

gbhall, 2015-08-09:
Sorry, here's the json generated by you:
```{"api":15,"build":"IML74K","cpu":["armeabi-v7a","armeabi"],"manufacturer":"HTC","product":"htc_pyramid","version":"4.0.3"}```

Further note, I see it's actually crashed the phones main camera app, I'm going to restart the device now...

commonsguy, 2015-08-09:
@gbhall It's very possible that my app's crashed tied up the camera, in a way that will mess up any other app, and that only a reboot can fix it. Thanks for the JSON!

gbhall, 2015-08-09:
@commonsguy Your app is working now after a a reboot. I'm thinking having it plugged into my PC may be the issue, as the SD card is mounted on the PC and unavailable to the device. Let me try that theory out again...

gbhall, 2015-08-09:
Plugging the device into the PC caused another error once I clicked the shutter. I'm not sure if you want it here or separate.

```
08-10 01:01:35.158    1931-2117/com.commonsware.cwac.cam2.demo E/AndroidRuntime﹕ FATAL EXCEPTION: pool-2-thread-1
    java.lang.UnsupportedOperationException: Exception when trying to write JPEG
            at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:99)
            at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
            at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:303)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1076)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:569)
            at java.lang.Thread.run(Thread.java:864)
     Caused by: java.io.FileNotFoundException: /cam2_HTC_htc_pyramid_20150810-010127/portrait-rear.jpg: open failed: ENOENT (No such file or directory)
            at libcore.io.IoBridge.open(IoBridge.java:406)
            at java.io.FileOutputStream.<init>(FileOutputStream.java:88)
            at java.io.FileOutputStream.<init>(FileOutputStream.java:73)
            at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:77)
            at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
            at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:303)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1076)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:569)
            at java.lang.Thread.run(Thread.java:864)
     Caused by: libcore.io.ErrnoException: open failed: ENOENT (No such file or directory)
            at libcore.io.Posix.open(Native Method)
            at libcore.io.BlockGuardOs.open(BlockGuardOs.java:110)
            at libcore.io.IoBridge.open(IoBridge.java:390)
            at java.io.FileOutputStream.<init>(FileOutputStream.java:88)
            at java.io.FileOutputStream.<init>(FileOutputStream.java:73)
            at com.commonsware.cwac.cam2.JPEGWriter.process(JPEGWriter.java:77)
            at com.commonsware.cwac.cam2.PictureTransaction.process(PictureTransaction.java:38)
            at com.commonsware.cwac.cam2.ClassicCameraEngine$TakePictureTransaction$1.run(ClassicCameraEngine.java:303)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1076)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:569)
            at java.lang.Thread.run(Thread.java:864)
```

commonsguy, 2015-08-09:
With regards to your second crash, it is incumbent on the calling app to confirm the availability of the storage location where it wants my code to write the output to. I have filed issue #72 to add this to the demo apps.

Let me know if you are able to reproduce the original crash. It doesn't look like I will be able to acquire that hardware, but for the moment I'm holding off worrying about it until you confirm that this is reproducible. Thanks!

gbhall, 2015-08-09:
@commonsguy I'll definitely try again! Thanks. I'm off to bed now as it's 1am here in Perth, Australia however I'll be back :)

gbhall, 2015-08-09:
It just happened again. I'll figure out the exact steps I'm taking to reproduce it.

commonsguy, 2015-09-05:
Closing as not reproducible. If you ever get this again and can give me steps to reproduce the crash, comment here and I can reopen the issue.

closed 2015-09-05 by commonsguy