commonsguy/cwac-cam2 #76: How to custom layout

shj00007, 2015-08-15:
I want to custom UI, can it do that?

gbhall, 2015-08-15:
Yes, I've done it and it's working perfectly :)

commonsguy, 2015-08-15:
@shj00007 There is no supported means of making a custom UI at this time, nor are there any plans for supporting a custom UI in the near future.

closed 2015-08-15 by commonsguy

guoyong0423, 2015-08-26:
I also want to custom layout...

markini, 2015-09-02:
@gbhall I guess you mean by forking the library? No?

gbhall, 2015-09-03:
@markini No need to fork. I did it by extending ```com.commonsware.cwac.cam2.CameraFragment``` and creating my own fragment.

VictorHVS, 2016-04-13:
@gbhall can you show us an example? How do you use the CameraActivity  with your custom fragment and call it on activity? thank you

VictorHVS, 2016-04-22:
@markini did you make this happening? XD Ty

ma2tkraus, 2016-04-26:
I too would find this very helpful. I've got a custom working version
(camera but no video) but it was painful to create and will be difficult to
support.

In my app the user can select the standard Android filesystem or IOCipher.
Ideally there is a callback that provides an OutputStream that can be
directed to a specific file.  The "protected" methods did not allow me to
call them directly. The design calls for a few obvious on-screen settings
and a full set of PreferenceActivity settings.

I found the Busy Coder's Guide quite helpful.

commonsguy, 2016-04-26:
@ma2tkraus Again, this is well outside the scope of what I want to handle with this library. If you need a custom UI, work with the camera APIs directly.

gbhall, 2016-05-05:
@VictorHVS Sorry, let me know if you still need a sample. I don't use it anymore though.

VictorHVS, 2016-05-05:
@gbhall I still need a sample, do you have any one? Ty very much! :)

gbhall, 2016-05-22:
@VictorHVS sorry for the massively long delay. I actually do not use this anymore, and this code was just a rough proof of concept, but it works :)

NOTE: This was last tested using `com.commonsware.cwac:cam2:0.4.3`

Fragment: http://pastebin.com/AAs4sQ0h
fragment_example_camera.xml: http://pastebin.com/ejaYB7zD

Whenever you want to use this custom camera with custom UI in your activity, just do:
```
mExampleCameraFragment = ExampleCameraFragment.newInstance();
getSupportFragmentManager().beginTransaction().replace(R.id.container, mExampleCameraFragment).commit();
```

laltukumar-1473, 2016-06-10:
@gbhall Have to tried upgrading your fragment using 0.5.0 version. Because camera controller parameters have changed in this version. If you example can you show me.