commonsguy/cwac-cam2 #82: Video Recording Not Stopping

commonsguy, 2015-09-05:
...or possibly not really starting. Pressing FAB to stop has no effect on FAB, no file shows up. Setting duration limit does not help. Reproducible on Nexus 5, Android 5.2 or 6.0 Final(?) Preview.

commonsguy, 2015-09-07:
No longer reproducible.

commonsguy, 2015-10-16:
Reproduced on Nexus 5 running Android 6.0.

commonsguy, 2015-10-16:
Also reproduced on a Nexus 4 running Android 5.1, but not on a Galaxy Nexus running Android 4.3. State of the force-classic switch has no effect.

bpappin, 2015-10-16:
I'm not sure if this is the same issue, but once started the video can't be stopped on a Samsung SGH-I337M.

This is the exception generated when he stop button is clicked.

``` 
10-16 12:06:04.267: W/MessageQueue(31006): Handler (android.os.Handler) {1bb94f4b} sending message to a Handler on a dead thread
10-16 12:06:04.267: W/MessageQueue(31006): java.lang.IllegalStateException: Handler (android.os.Handler) {1bb94f4b} sending message to a Handler on a dead thread
10-16 12:06:04.267: W/MessageQueue(31006): 	at android.os.MessageQueue.enqueueMessage(MessageQueue.java:325)
10-16 12:06:04.267: W/MessageQueue(31006): 	at android.os.Handler.enqueueMessage(Handler.java:631)
10-16 12:06:04.267: W/MessageQueue(31006): 	at android.os.Handler.sendMessageAtFrontOfQueue(Handler.java:623)
10-16 12:06:04.267: W/MessageQueue(31006): 	at android.hardware.camera2.legacy.RequestThreadManager.quit(RequestThreadManager.java:941)
10-16 12:06:04.267: W/MessageQueue(31006): 	at android.hardware.camera2.legacy.LegacyCameraDevice.close(LegacyCameraDevice.java:444)
10-16 12:06:04.267: W/MessageQueue(31006): 	at android.hardware.camera2.legacy.LegacyCameraDevice.finalize(LegacyCameraDevice.java:468)
10-16 12:06:04.267: W/MessageQueue(31006): 	at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:194)
10-16 12:06:04.267: W/MessageQueue(31006): 	at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:177)
10-16 12:06:04.267: W/MessageQueue(31006): 	at java.lang.Thread.run(Thread.java:818)
```

commonsguy, 2015-10-17:
@bpappin What version of Android is your SGH-I337M running? If it's 4.4 or earlier, you have a separate problem.

commonsguy, 2015-10-17:
For Android 5.0 and higher, until issue #57 is addressed, we have to use the classic camera engine, and I misplaced my code to force Android 5.0+ devices to use that. This will be addressed in this weekend's update.

commonsguy, 2015-10-18:
Addressed in v0.3.2.

closed 2015-10-18 by commonsguy