commonsguy/cwac-cam2 #85: How to use CameraFragment without CameraActivity?

zokipirlo, 2015-09-07:
Hi!

In docs there is only an option to start camera activity for result. But I have an Activity with bottom tabs, similar to Facebook Messenger.
I tried to show CameraFragment inside holder ... it worked, but I only see never ending loader. I guess that it needs some event handling. Is it possible to implement that?

Thanks!

commonsguy, 2015-09-07:
Quoting [the documentation](https://github.com/commonsguy/cwac-cam2#basic-usage):

> The only supported API at the moment is through CameraActivity and its IntentBuilder for still pictures and VideoRecorderActivity and its IntentBuilder for videos.

> While there are other public classes and methods in the library, ones that may be exposed as part of a public API in the future, they are not supported at present.

So, at the present time, please use the Android camera APIs directly.

closed 2015-09-07 by commonsguy

zokipirlo, 2015-09-08:
Ok, thanks. Is there any plan for that?

commonsguy, 2015-09-08:
@zokipirlo I will be astonished if it ever happens. I tried to provide a more flexible library with the original CWAC-Camera, and that flexibility contributed to a lot of problems that I had with the library. My response to that is to focus on reliability and stability first, and worry about flexibility somewhere down the road.

That being said, please understand that I am not stopping you from using the classes like `CameraFragment`. This is an open source project, after all. I'm just not supporting their use in that way, in terms of documentation, answers to questions, design changes, etc. You are certainly welcome to file issues here (bugs, feature requests, etc.) that you run into when using those classes, so long as those issues *also* affect the library's official API and the functionality exposed by that API.

BTW, for questions, please use Stack Overflow, as is mentioned in [the documentation](https://github.com/commonsguy/cwac-cam2#questions), so we can reserve the issue tracker for issues. Thanks!

zokipirlo, 2015-09-08:
Thanks again, I missed that.

zokipirlo, 2015-09-29:
Hi again :)

I have implemented this. I had to do some refactoring (ids in xml for skinning, replaced fab with imagebutton (in code only, because fab extends imagebutton), added expanded button, some small refactoring in code, added support library for nested fragments).

I would be very grateful if we could merge that somehow, if you're interested in this of course. It doesn't break your code.

Here are some screenshots.

Before capture:
[image]

Preview with option for retake:
[image]

Expanded (full screen):
[image]

commonsguy, 2015-09-29:
> I would be very grateful if we could merge that somehow

I have no idea what "that" is, sorry.

zokipirlo, 2015-09-29:
I was referencing with "that" on an original question. It adds an option to use FullCameraFragment inside any view. Like it's visible in screenshots above (it's hard to see, but I didn't find an option to make them smaller :( ).
I have also added an example in demo-playground project. It's implemented in FragmentDemo activity (Cam2 Fragment Demo in launcher) .
It's pushed now in my fork (https://github.com/zokipirlo/cwac-cam2).

commonsguy, 2015-09-29:
I have no plans to offer an API other than the activity-based one. If you would like, I can steer people looking for a fragment-based API to your fork.

If there are trivial changes that I can make to the library that will help you in maintaining your fork, akin to issue #94, feel free to file issues for them.

Or, if you are invested in writing an open source library for simplifying taking pictures within an Android app, we can invert the relationship and make your fork the "real one" that people use, with me perhaps contributing bits and pieces along the way. I need such a library to exist, but it's not essential to me that *I* be the one to write the library. I am writing this library, and wrote its predecessor, to scratch a particular itch, but that itch can be just as easily scratched via somebody else's library.

zokipirlo, 2015-09-30:
Ok, great, thanks. Will start a fork from scratch and create few small compatibility issues with pull requests.
I'm planning to use this library in a commercial application, so I will regularly update my fork. Would be glad if you continue to support this fork. Thanks.

gepriniusce, 2016-03-01:
I also want to customize my camera view similar to Instagram  .I used to CWAC-Camera,but the flash light doesn't work.Should I rewrite the CameraFragment according to cam2 libs?

commonsguy, 2016-03-01:
@gepriniusce Use the camera APIs directly. Do not use this library.