commonsguy/cwac-cam2 #89: VideoTansaction.Builder methods should be public

gryffindor14, 2015-09-08:
Name says it all. If we try and use the transactions outside of the VideoActivity we can't use the build method to even create it

commonsguy, 2015-09-08:
@gryffindor14: Quoting [the documentation](https://github.com/commonsguy/cwac-cam2#basic-usage):

> The only supported API at the moment is through CameraActivity and its IntentBuilder for still pictures and VideoRecorderActivity and its IntentBuilder for videos.

> While there are other public classes and methods in the library, ones that may be exposed as part of a public API in the future, they are not supported at present.

`VideoTransaction`, therefore, is not part of the public API. I don't mind making fairly trivial changes like this, but please understand that what you are doing is not supported.

commonsguy, 2015-09-13:
Addressed in v0.2.2.

closed 2015-09-13 by commonsguy