commonsguy/cwac-cam2 #90: Orientation issues on Samsung Galaxy S4

andrewprocter, 2015-09-08:
MANUFACTURER: samsung 
PRODUCT:jfltevl

Seems like photos taken in portrait are always returned rotated 90 degrees when returned in onActivityResult. I would attach a zip of the demo app output, but the images are fine there. The behaviour can be seen in demo-playground on this device though. Screenshot:

[image]

commonsguy, 2015-09-08:
Most likely, this is a duplicate of issue #15. Examine the taken photo and see if if has an EXIF orientation header. If it does, then that's what issue #15 is for. If not, it may be a duplicate of issue #45.

andrewprocter, 2015-09-08:
Ah, they do have the orientation header: Orientation: 8 (Rotated 90° CW)

commonsguy, 2015-09-08:
Yeah, that'll be addressed by issue #15. One of the many aggravations of working with the camera APIs.

Closing as a duplicate of issue #15.

closed 2015-09-08 by commonsguy

andrewprocter, 2015-09-08:
I've subscribed to notifications on #15! Thanks for all your work on this.