commonsguy/cwac-cam2 #91: Floating Action Button icons don't switch properly.

andrewprocter, 2015-09-08:
Using the default UI, the FAB settings cog icon does not switch to a close icon when pressed for the first time. It does switch to the close icon when tapped again, resulting in the state being backwards - it is now a close button until opened, and a settings cog until closed. Seems like this is a boolean being set wrong by default, maybe in onCreateView in CameraFragment.java?

[image]
[image]

commonsguy, 2015-09-08:
Yes, something's up here. I can reproduce this on a Nexus 5 running Android 5.1.

Thanks!

andrewprocter, 2015-09-08:
I'm wondering - is the settings cog holding the flip camera button necessary, since it is the only action in there? Could it just be replaced with the flip camera button? Admittedly I haven't tried video mode and have only tested on Nexus 5 and Samsung S4, so if it holds more than one action sometimes I just haven't seen it.

commonsguy, 2015-09-08:
There will be other settings in time (flash mode, etc.). Knowing that, I went ahead and set up a FAM (floating action menu) with that in mind. Note that the UI there is not set in stone -- I may wind up switching to an action sheet or one of the other material-ish patterns -- but there will be several peer options eventually.

commonsguy, 2015-09-13:
Addressed in v0.2.2.

closed 2015-09-13 by commonsguy