commonsguy/cwac-cam2 #94: Include PictureTransaction in PictureTakenEvent to match the VideoTakenEvent (enhancement)

gryffindor14, 2015-09-16:
I understand this is unsupported but it makes sense to match. 
CameraEngine.Java

public static class PictureTakenEvent extends CameraEngine.CrashableEvent {
        private ImageContext imageContext = null;
        private PictureTransaction xact = null;
        public PictureTakenEvent(PictureTransaction xact, ImageContext imageContext) {
            this.imageContext = imageContext;
            this.xact = xact;
        }
        public PictureTakenEvent(Exception exception) {
            super(exception);
        }
        public ImageContext getImageContext() {
            return this.imageContext;
        }
       public PictureTrasaction getPictureTransaction(){
           return xact;
    }
ClassicCameraEngine.java

private class TakePictureTransaction implements Camera.PictureCallback {
    private final PictureTransaction xact;
    private final Context ctxt;

    TakePictureTransaction(Context ctxt, PictureTransaction xact) {
      this.ctxt=ctxt.getApplicationContext();
      this.xact=xact;
    }

    @Override
    public void onPictureTaken(final byte[] bytes, final Camera camera) {
      getThreadPool().execute(new Runnable() {
        @Override
        public void run() {
          camera.startPreview();
          getBus().post(new PictureTakenEvent(xact, xact.process(new ImageContext(ctxt, bytes))));
        }
      });
    }
  }

gryffindor14, 2015-09-16:
CameraTwoEngine 

 private static class TakePictureTransaction implements ImageReader.OnImageAvailableListener {
    private final EventBus bus;
    private final PictureTransaction xact;
    private final Context ctxt;

    TakePictureTransaction(Context ctxt, EventBus bus, PictureTransaction xact) {
      this.bus=bus;
      this.xact=xact;
      this.ctxt=ctxt.getApplicationContext();
    }

    @Override
    public void onImageAvailable(ImageReader imageReader) {
      Image image=imageReader.acquireNextImage();
      ByteBuffer buffer=image.getPlanes()[0].getBuffer();
      byte[] bytes=new byte[buffer.remaining()];

      buffer.get(bytes);
      image.close();

      bus.post(new PictureTakenEvent(xact, xact.process(new ImageContext(ctxt, bytes))));
    }
  }

commonsguy, 2015-09-16:
Um, I'm confused. What are you talking about?

gryffindor14, 2015-09-16:
The VideoTransactionEvent class has a reference to the VideoTransaction in it. The PictureTransactionEvent class does not have a reference to PictureTransaction. It only has the ImageContext. I understand it isn't supported in your use case but I need to get the output Path from the Picture transaction in the same way I can get the output path from the Video transaction. It makes sense to match the VideoTransactionEvent as well

commonsguy, 2015-09-26:
Addressed in v0.3.0.

closed 2015-09-26 by commonsguy