commonsguy/cwac-cam2 #99: demo-playground record video lag before record and stop

bgrayburn, 2015-09-23:
Thanks for this great library, super handy! I've tried using the demo-playground app on a nexus 5 and a moto g, with some problems. When you click the record button the button turns to the red stop button which seems to make sense, but there is no output on `adb logcat | grep QCam` following the record button press. When I press the red stop button that now is visible, nothing at all happens, the button doesn't change and again no output on the above `logcat`. Any help appreciated, let me know if you need more info.

commonsguy, 2015-09-23:
I just ran the current video playground app on a Nexus 5, itself running Android 5.1.1, without problems (other than it attempting to launch a non-existent video player after recording the video). The playground settings were:

- Use front-facing camera: off
- Update MediaStore: off
- Enable debug mode: off
- Force classic API: on
- Record high-quality video: on
- No duration or size limits

> there is no output on adb logcat | grep QCam following the record button press

Since I did not write any log messages containing "QCam", I have no way to control whether such messages appear or not. I don't even know what "QCam" is.

> When I press the red stop button that now is visible, nothing at all happens, the button doesn't change

If you are able to reproduce the problem on a Nexus 5, please indicate the version of Android that is on it and what your playground settings are, so that I can try to reproduce it here.

Thanks!

bgrayburn, 2015-09-23:
ok I found part of the problem, having adb running is what was creating
some of the slowness/unresponsiveness I saw earlier

I'm on Android ver 5.1.1 (nexus 5)
and Android ver 5.0.2 (moto g)
btw the photo playground app works fine on both

here's what's happening (and going wrong) now:

on the nexus 5:
With the settings you provided, the app now makes an audible beep when I
click record and stop, which I'm assuming is a good sign, this wasn't
happening before.  Unfortunately it's also crashing as soon as I hit stop,
with the os giving me the error `Unfortunately, Cam2 Picture Playground has
stopped.`

on the moto g:
same as the nexus 5 except still no beeping

commonsguy, 2015-09-23:
> Unfortunately it's also crashing as soon as I hit stop, with the os giving me the error `Unfortunately, Cam2 Picture Playground has stopped.`

If you look at the stack trace, it probably is complaining about a lack of a video player, as I am trying to launch one to play back the just-recorded video. See issue #86.

> same as the nexus 5 except still no beeping

Whether or not there is beeping is a device-specific thing. What matters is whether a video was recorded, to `getExternalFilesDir()` as `temp.mp4`. Check to see if there is such a video file.

bgrayburn, 2015-09-23:
found the video in the filesystem, didn't know how to check the stack trace
without adb, but it looks like you were right on. Thank you!

closed 2015-09-23 by bgrayburn