commonsguy/cwac-cam2 #106: Nexus 9: Auto-Focus Messed Up for RFC

commonsguy, 2015-10-04:
Run demo app. Tap FAB to take picture. Watch as the auto-focus rapid-fire goes in and out. Add hack for this, to disable auto-focus, at least on 5.1.1 (retest on 6.0 once upgrade rolls around, as the M Developer Preview does not exhibit this behavior).

markini, 2015-10-15:
I think I have the same issue on 6.0. When taking the photo the auto-focus makes the preview very blurry (depending on light and distance --> it is definitively worse than every other device I have at my disposal).

N9, 6.0, MRA58K

commonsguy, 2015-10-18:
Addressed in v0.3.2.

markini, 2015-10-27:
Oh, my device has 
`Build.MANUFACTURER` = `HTC`
`Build.PRODUCT` = `VOLANTISG`

commonsguy, 2015-10-27:
@markini Just to confirm -- those are both in all caps?

markini, 2015-10-28:
Oh sorry, don't know how this ended up in all caps (had an app which printed it all caps). It's all in lowercase letters. 
Here a dalvik explorer screenshot:

[image]

commonsguy, 2015-11-01:
Addressed in v0.3.4.

closed 2015-11-01 by commonsguy