commonsguy/cwac-cam2 #107: CWAC - cam 2 , No camera preview, only loading bar

Parveen-okido, 2015-10-15:
Intent captureIntent = new CameraActivity.IntentBuilder(this)
                .skipConfirm().debug().to(captureImageFile).build();

        startActivityForResult(captureIntent, REQUEST_PORTRAIT_FFC);

When i run this code, it starts camera activity with loading bar, No camera preview is there.
2 - 3 days before it was working fine.
Thnx

commonsguy, 2015-10-15:
Does this occur with either of the demo apps? If yes, please indicate which app, what you did to reproduce the problem, and what hardware you are testing on (`Build.MANUFACTURER` and `Build.PRODUCT` values).

If this does not occur with the demo apps, I would need the source code of an app that can reproduce the problem, along with the above information.

Parveen-okido, 2015-10-15:
i didn't check demo app.
I am using Multiple image Picker Lib(Pic images from gallery), and inside there i am using cwac-cam2.
I tested on nexus 7 tablet, Samsung Duos, Micromax A114.

  public void capturePhoto() {
        final File captureImageFile = createTemporaryFileForCapturing(".png");
        final Intent captureIntent = new CameraActivity.IntentBuilder(this)
                .skipConfirm()
                .debug()
                .to(captureImageFile)
                .build();
        startActivityForResult(captureIntent, REQUEST_PORTRAIT_FFC);
    }
    private File createTemporaryFileForCapturing(final String extension) {
        final File captureTempFile = new File(CAPTURED_IMAGES_DIR + "/tmp" + System.currentTimeMillis() + extension);
        try {
            captureTempFile.createNewFile();
        } catch (IOException e) {
            e.printStackTrace();
            Log.e("capture", e.getMessage());
        }
        return captureTempFile;
    }

Please check attached log.txt file.

Could not find method com.commonsware.cwac.cam2.AbstractCameraActivity.checkSelfPermission, referenced from method com.commonsware.cwac.cam2.AbstractCameraActivity.hasPermission

VFY: unable to resolve virtual method 19100: Lcom/commonsware/cwac/cam2/AbstractCameraActivity;.checkSelfPermission (Ljava/lang/String;)I

Parveen-okido, 2015-10-15:
[image]

Logcat Output:

10-15 20:01:26.871: D/ActivityThread(29383): ACT-AM_ON_PAUSE_CALLED ActivityRecord{42192088 token=android.os.BinderProxy@422327d8 {com.okido.school/com.okido.school.imagepicker.ui.PickerActivity}}
10-15 20:01:26.999: D/ActivityThread(29383): ACT-PAUSE_ACTIVITY handled : 1 / android.os.BinderProxy@422327d8
10-15 20:01:27.044: D/SystemTheme(29580): sThemeResIdMap = {16974129=16974315, 16974128=16974314, 16974135=16974317, 16974134=16974316, 16973932=16974318, 16973933=16974319, 16973934=16974311, 16973935=16974312, 16974619=16974313, 16973931=16974311, 16974631=16974313, 16973943=16974316, 16973937=16974314, 16973936=16974322, 16973938=16974315, 16974120=16974311, 16974126=16974312, 16974127=16974322, 16973944=16974317, 16973947=16974321}
10-15 20:01:27.046: D/SystemTheme(29580): sPackageList = [android, com.android.gallery3d, com.android.settings, com.mediatek.voiceunlock, com.android.phone, com.android.providers.media, com.android.inputmethod.latin, com.android.systemui, com.android.browser, com.android.packageinstaller, com.mediatek.bluetooth, com.android.providers.downloads.ui, com.android.stk, com.android.music, com.mediatek.schpwronoff, com.android.contacts, com.mediatek.CellConnService, com.android.calendar, com.mediatek.videoplayer, com.mediatek.security, com.mediatek.ppl, com.mediatek.op01.plugin]
10-15 20:01:27.053: D/GraphicBuffer(29383): close handle(0x60faef70) (w:608 h:976 f:2)
10-15 20:01:27.055: D/GraphicBuffer(29383): close handle(0x60fb2c70) (w:608 h:976 f:2)
10-15 20:01:27.057: V/Provider/Settings(29580): invalidate [system]: current 322 != cached 0
10-15 20:01:27.061: D/GraphicBuffer(29383): close handle(0x6098b860) (w:608 h:976 f:2)
10-15 20:01:27.063: D/ActivityThread(29580): hoder:android.app.IActivityManager$ContentProviderHolder@41c8c190,provider,holder.Provider:android.content.ContentProviderProxy@41c8c908
10-15 20:01:27.065: V/Provider/Settings(29580): from db cache, name = anr_debugging_mechanism , value = 1
10-15 20:01:27.069: D/asset(29580): AssetManager-->addDefaultAssets CIP path not exsit!
10-15 20:01:27.078: D/Proxy(29580): setHttpRequestCheckHandler
10-15 20:01:27.080: D/dalvikvm(29580): open_cached_dex_file : /data/app/com.okido.school-1.apk /data/dalvik-cache/data@app@com.okido.school-1.apk@classes.dex
10-15 20:01:27.092: D/dalvikvm(29580): create interp thread : stack size=128KB
10-15 20:01:27.092: D/dalvikvm(29580): create new thread
10-15 20:01:27.093: D/dalvikvm(29580): new thread created
10-15 20:01:27.093: D/dalvikvm(29580): update thread list
10-15 20:01:27.093: D/dalvikvm(29580): threadid=11: interp stack at 0x60764000
10-15 20:01:27.093: D/dalvikvm(29580): threadid=11: created from interp
10-15 20:01:27.093: D/dalvikvm(29580): start new thread
10-15 20:01:27.093: D/dalvikvm(29580): threadid=11: notify debugger
10-15 20:01:27.093: D/dalvikvm(29580): threadid=11 (AsyncTask #1): calling run()
10-15 20:01:27.094: D/dalvikvm(29580): create interp thread : stack size=128KB
10-15 20:01:27.094: D/dalvikvm(29580): create new thread
10-15 20:01:27.095: D/dalvikvm(29580): new thread created
10-15 20:01:27.095: D/dalvikvm(29580): update thread list
10-15 20:01:27.095: D/dalvikvm(29580): threadid=12: interp stack at 0x60882000
10-15 20:01:27.095: D/dalvikvm(29580): threadid=12: created from interp
10-15 20:01:27.095: D/dalvikvm(29580): start new thread
10-15 20:01:27.095: D/dalvikvm(29580): threadid=12: notify debugger
10-15 20:01:27.095: D/dalvikvm(29580): threadid=12 (SharedPreferencesImpl-load): calling run()
10-15 20:01:27.097: D/dalvikvm(29580): create interp thread : stack size=128KB
10-15 20:01:27.097: D/dalvikvm(29580): create new thread
10-15 20:01:27.097: D/dalvikvm(29580): new thread created
10-15 20:01:27.097: D/dalvikvm(29580): update thread list
10-15 20:01:27.097: D/dalvikvm(29580): threadid=13: interp stack at 0x609a0000
10-15 20:01:27.097: D/dalvikvm(29580): threadid=13: created from interp
10-15 20:01:27.097: D/dalvikvm(29580): start new thread
10-15 20:01:27.098: D/dalvikvm(29580): threadid=13: notify debugger
10-15 20:01:27.098: D/dalvikvm(29580): threadid=13 (AsyncTask #2): calling run()
10-15 20:01:27.099: D/dalvikvm(29580): create interp thread : stack size=128KB
10-15 20:01:27.099: D/dalvikvm(29580): create new thread
10-15 20:01:27.099: D/dalvikvm(29580): new thread created
10-15 20:01:27.099: D/dalvikvm(29580): update thread list
10-15 20:01:27.100: D/dalvikvm(29580): threadid=14: interp stack at 0x60abe000
10-15 20:01:27.100: D/dalvikvm(29580): threadid=14: created from interp
10-15 20:01:27.100: D/dalvikvm(29580): start new thread
10-15 20:01:27.103: D/dalvikvm(29580): create interp thread : stack size=128KB
10-15 20:01:27.103: D/dalvikvm(29580): create new thread
10-15 20:01:27.104: D/dalvikvm(29580): new thread created
10-15 20:01:27.104: D/dalvikvm(29580): update thread list
10-15 20:01:27.104: D/dalvikvm(29580): threadid=14: notify debugger
10-15 20:01:27.104: D/dalvikvm(29580): threadid=14 (AsyncTask #3): calling run()
10-15 20:01:27.104: D/dalvikvm(29580): threadid=15: interp stack at 0x60bdc000
10-15 20:01:27.104: D/dalvikvm(29580): threadid=15: created from interp
10-15 20:01:27.105: D/dalvikvm(29580): start new thread
10-15 20:01:27.105: D/ActivityThread(29580): BIND_APPLICATION handled : 0 / AppBindData{appInfo=ApplicationInfo{41c82a20 com.okido.school}}
10-15 20:01:27.105: D/dalvikvm(29580): threadid=15: notify debugger
10-15 20:01:27.106: D/dalvikvm(29580): threadid=15 (AsyncTask #4): calling run()
[... cut ...]
10-15 20:01:28.134: D/dalvikvm(29580): threadid=17: created from interp
10-15 20:01:28.134: D/dalvikvm(29580): start new thread
10-15 20:01:28.135: D/dalvikvm(29580): threadid=17: notify debugger
10-15 20:01:28.135: D/dalvikvm(29580): threadid=17 (pool-4-thread-1): calling run()
10-15 20:02:22.800: D/dalvikvm(29383): threadid=12: exiting
10-15 20:02:22.800: D/dalvikvm(29383): threadid=12: bye!
10-15 20:02:28.127: D/dalvikvm(29580): threadid=16: exiting
10-15 20:02:28.127: D/dalvikvm(29580): threadid=16: bye!

commonsguy, 2015-10-15:
Your quoted messages are not errors and will appear on any Android app, running on a pre-Android 6.0 device, that has references to the new runtime permissions methods.

> i didn't check demo app.

Please do. I need to have all the pieces necessary to try to reproduce the problem. Otherwise, there is nothing that I can do.

> I tested on nexus 7 tablet, Samsung Duos, Micromax A114.

There are two different Nexus 7 models and perhaps a dozen or more Samsung devices with "Duos" in the name. With respect to the Nexus 7, does yours have a rear-facing camera (2013 model) or not (2012 model)?

Parveen-okido, 2015-10-15:
ASUS Nexus 7 Model ME370T, OS 5.1.1, Build Number LMY47V. (Only Front Camera)
Samsung Model No. GT-I8262, OS 4.1.2, 
Lenovo TAB2 A7-30HC, OS 4.4.2

I am getting error in demo app.
@ line  Camera camera=Camera.open(descriptor.getCameraId()); when i press 'Capture' button
10-15 20:15:35.404: W/System.err(29924): 	at android.hardware.Camera.native_setup(Native Method)
10-15 20:15:35.404: W/System.err(29924): 	at android.hardware.Camera.<init>(Camera.java:469)
10-15 20:15:35.404: W/System.err(29924): 	at android.hardware.Camera.open(Camera.java:424)
10-15 20:15:35.404: W/System.err(29924): 	at com.commonsware.cwac.cam2.ClassicCameraEngine$1.run(ClassicCameraEngine.java:66)
10-15 20:15:35.404: W/System.err(29924): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
10-15 20:15:35.404: W/System.err(29924): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
10-15 20:15:35.404: W/System.err(29924): 	at java.lang.Thread.run(Thread.java:841)
10-15 20:15:35.404: W/dalvikvm(29924): threadid=11: calling UncaughtExceptionHandler
10-15 20:15:35.405: E/AndroidRuntime(29924): FATAL EXCEPTION: pool-2-thread-1
10-15 20:15:35.405: E/AndroidRuntime(29924): Process: com.commonsware.cwac.cam2.demo:cwac_cam2, PID: 29924
10-15 20:15:35.405: E/AndroidRuntime(29924): java.lang.RuntimeException: Fail to connect to camera service
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at android.hardware.Camera.native_setup(Native Method)
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at android.hardware.Camera.<init>(Camera.java:469)
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at android.hardware.Camera.open(Camera.java:424)
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at com.commonsware.cwac.cam2.ClassicCameraEngine$1.run(ClassicCameraEngine.java:66)
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
10-15 20:15:35.405: E/AndroidRuntime(29924): 	at java.lang.Thread.run(Thread.java:841)
10-15 20:15:35.438: D/ActivityThread(29924): ACT-AM_ON_RESUME_CALLED ActivityRecord{41c8cf18 token=android.os.BinderProxy@41c8c628 {com.commonsware.cwac.cam2.demo/com.commonsware.cwac.cam2.CameraActivity}}
10-15 20:15:35.439: V/PhoneWindow(29924): DecorView setVisiblity: visibility = 4 ,Parent =null, this =com.android.internal.policy.impl.PhoneWindow$DecorView{41c9c4c8 I.E..... R.....ID 0,0-0,0}
10-15 20:15:35.447: V/PhoneWindow(29924): DecorView setVisiblity: visibility = 0 ,Parent =ViewRoot{41cfaef0 com.commonsware.cwac.cam2.demo/com.commonsware.cwac.cam2.CameraActivity,ident = 0}, this =com.android.internal.policy.impl.PhoneWindow$DecorView{41c9c4c8 V.E..... R.....ID 0,0-0,0}

commonsguy, 2015-10-18:
I cannot reproduce your problem with the Nexus 7 (from your last comment), running the same hardware with the same build. The demo app takes the portrait and landscape pictures without issue, tested from a cold boot.

I have no means of reproducing your original problem. If, at some point, you wish to publish a project that reproduces that problem, comment back here and I can reopen the issue.

But, for now, I am closing this as not reproducible.

closed 2015-10-18 by commonsguy

Parveen-okido, 2015-10-19:
Finally i solved it !!.
Problem was in Manifest file.
i changed  android:hardwareAccelerated="false" to android:hardwareAccelerated="true", and now its working fine !!
I wasted 2 days on this :|

commonsguy, 2015-10-19:
@Parveen-okido Ah, I should have thought of that. My apologies. I have added a note to the `README` about hardware acceleration.

Parveen-okido, 2015-10-20:
Hi, I just want to know that how can i manage captured image size. I am getting 3 - 4 MB image from camera. I have to upload images to server, from where  i can keep image size around 500 kb.

commonsguy, 2015-10-20:
@Parveen-okido Please use Stack Overflow for general questions, as is described in [the project `README`](https://github.com/commonsguy/cwac-cam2#questions).

You will need to scale the images yourself, whether using the NDK or using something like `BitmapFactory.Options` and `inSampleSize`.

Parveen-okido, 2015-10-20:
Ok i made some changes in "JPEGWriter.java" and its working fine.

           FileOutputStream fos=new FileOutputStream(f);
          Bitmap bitmap = imageContext.getBitmap(true);
          ByteArrayOutputStream bytes = new ByteArrayOutputStream();
         // fos.write(imageContext.getJpeg());
          bitmap.compress(Bitmap.CompressFormat.JPEG, 50, bytes);
          fos.write(bytes.toByteArray());
          fos.flush();
          fos.getFD().sync();
          fos.close();

Original image size was 2.94 MB, now its 343 KB.