commonsguy/cwac-cam2 #108: SONY Xperia Z3 Compact: Autofocus not working

markini, 2015-10-15:
Hello, I can't take pictures on the Sony Z3 compact (I can give you the exact specifications tomorrow, when I have access to it again).

Log output:
```java
10-15 17:15:21.773 7181-7181/at.mytestapp.test I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@941138 time:120842439
10-15 17:15:24.218 7181-7181/at.mytestapp.test I/Timeline: Timeline: Activity_launch_request id:at.mytestapp.test time:120844884
10-15 17:15:24.324 7181-8303/at.mytestapp.test I/CameraManager: Using legacy camera HAL.
10-15 17:15:24.375 7181-7430/at.mytestapp.test D/OpenGLRenderer: endAllStagingAnimators on 0xa0729500 (RippleDrawable) with handle 0x9e112290
10-15 17:15:24.379 7181-7181/at.mytestapp.test I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@2c73ed5e time:120845045
10-15 17:15:24.589 7181-8302/at.mytestapp.test I/CameraDeviceState: Legacy camera service transitioning to state CONFIGURING
10-15 17:15:24.591 7181-12397/at.mytestapp.test I/RequestThread-0: Configure outputs: 2 surfaces configured.
10-15 17:15:24.591 7181-12397/at.mytestapp.test D/Camera: app passed NULL surface
10-15 17:15:24.594 7181-12397/at.mytestapp.test I/RequestThread-0: configureOutputs - set take picture size to 5248x3936
10-15 17:15:24.629 7181-8302/at.mytestapp.test I/CameraDeviceState: Legacy camera service transitioning to state IDLE
10-15 17:15:24.636 7181-8302/at.mytestapp.test E/commonsware: desiredMode=4
10-15 17:15:24.637 7181-8302/at.mytestapp.test E/commonsware: itsAllGood=true
10-15 17:15:24.637 7181-8302/at.mytestapp.test I/RequestQueue: Repeating capture request set.
10-15 17:15:24.640 7181-12397/at.mytestapp.test W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
10-15 17:15:24.640 7181-12397/at.mytestapp.test W/LegacyRequestMapper: Only received metering rectangles with weight 0.
10-15 17:15:25.011 7181-12398/at.mytestapp.test I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
10-15 17:15:27.124 7181-8303/at.mytestapp.test I/RequestQueue: Repeating capture request cancelled.
10-15 17:15:27.124 7181-8303/at.mytestapp.test I/RequestQueue: Repeating capture request set.
10-15 17:15:27.140 7181-12397/at.mytestapp.test W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
10-15 17:15:27.140 7181-12397/at.mytestapp.test W/LegacyRequestMapper: Only received metering rectangles with weight 0.
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0: Received device exception during capture call: 
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0: java.lang.RuntimeException: autoFocus failed
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.hardware.Camera.native_autoFocus(Native Method)
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.hardware.Camera.autoFocus(Camera.java:1299)
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.hardware.camera2.legacy.LegacyFocusStateMapper.processRequestTriggers(LegacyFocusStateMapper.java:205)
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.hardware.camera2.legacy.RequestThreadManager$5.handleMessage(RequestThreadManager.java:808)
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.os.Handler.dispatchMessage(Handler.java:98)
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.os.Looper.loop(Looper.java:211)
10-15 17:15:27.170 7181-12397/at.mytestapp.test E/RequestThread-0:     at android.os.HandlerThread.run(HandlerThread.java:61)
10-15 17:15:27.170 7181-12397/at.mytestapp.test I/CameraDeviceState: Legacy camera service transitioning to state ERROR
10-15 17:15:27.196 7181-12398/at.mytestapp.test I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
10-15 17:15:27.196 7181-12398/at.mytestapp.test E/CameraDeviceState: Cannot call capture while in state: 0
10-15 17:15:27.217 7181-12398/at.mytestapp.test W/Adreno-EGL: <qeglDrvAPI_eglDestroyImageKHR:4981>: failed to wait for 0xaeeb5f40 syncobj
10-15 17:15:27.217 7181-12398/at.mytestapp.test W/Adreno-EGL: <qeglDrvAPI_eglDestroyImageKHR:4981>: failed to wait for 0xaed95b08 syncobj
10-15 17:15:27.244 7181-7206/at.mytestapp.test E/BufferQueueProducer: [unnamed-7181-14] dequeueBuffer: BufferQueue has been abandoned
10-15 17:15:27.246 7181-7204/at.mytestapp.test E/BufferQueueProducer: [unnamed-7181-14] cancelBuffer: BufferQueue has been abandoned
```

I'm pretty convinced that it is the same issue as: #96

commonsguy, 2015-10-15:
@markini It's very possible that this is #96. If you can get me the `Build.MANUFACTURER` and `Build.PRODUCT` values, I can patch it, pending my eventually getting my hands on one of these things. Thanks!

markini, 2015-10-19:
`Build.MANUFACTURER` = `SONY`
`Build.PRODUCT` = `D5803`

commonsguy, 2015-10-19:
@markini OK, in my next pulse on this, I'll add in code to detect this, and we can see if that helps. Thanks!

commonsguy, 2015-10-25:
Addressed in v0.3.3. Let me know if this clears up your problem.

closed 2015-10-25 by commonsguy