commonsguy/cwac-cam2 #109: Can't start VideoRecorderActivity

bpappin, 2015-10-16:
When trying to launch the VideoRecorderActivity, i get this exception:

```
10-16 11:01:16.914: E/AndroidRuntime(26674): Caused by: de.greenrobot.event.EventBusException: Subscriber class com.commonsware.cwac.cam2.CameraController has no public methods called onEvent
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at de.greenrobot.event.SubscriberMethodFinder.findSubscriberMethods(SubscriberMethodFinder.java:119)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at de.greenrobot.event.EventBus.register(EventBus.java:163)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at de.greenrobot.event.EventBus.register(EventBus.java:133)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at com.commonsware.cwac.cam2.CameraController.setEngine(CameraController.java:76)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at com.commonsware.cwac.cam2.AbstractCameraActivity.init(AbstractCameraActivity.java:304)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at com.commonsware.cwac.cam2.AbstractCameraActivity.onCreate(AbstractCameraActivity.java:186)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at android.app.Activity.performCreate(Activity.java:6289)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1119)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2646)
10-16 11:01:16.914: E/AndroidRuntime(26674): 	... 10 more

```

commonsguy, 2015-10-16:
Please provide steps to reproduce this problem, as I cannot reproduce it using the `demo-playground` sample app.

bpappin, 2015-10-16:
I simply call, and I'm using the most recent EventBus library listed on their site.
It looks like the controller is registering itself, doubt doesn't have a way to receive events.

I am checking out the cam2 code to see if I can track down he exact issue.

``` java

VideoRecorderActivity.IntentBuilder b = new VideoRecorderActivity.IntentBuilder(activity);
		File tempFile = File.createTempFile("video_", "_cx2s.mp4", activity.getExternalCacheDir());
		b.to(tempFile);
		b.quality(VideoRecorderActivity.Quality.LOW);
		b.facing(AbstractCameraActivity.Facing.FRONT);
		if (Const.CAMERA_DEBUG) {
			b.debug();
		}
		b.durationLimit(Const.Limits.DURATION_LIMIT_MILLIS);
		b.sizeLimit(Limits.BYTE_LIMIT);
		activity.startActivityForResult(b.build(), Result.REQUEST_CODE_CAPTURE_VIDEO);

```

commonsguy, 2015-10-16:
If you are using ProGuard, please do not obfuscate the cam2 classes, or at least ensure that all methods beginning with `onEvent...()` are kept intact.

bpappin, 2015-10-16:
I'm build a debug build which pro guard doesn't run against, however thats still an important  point. I don't see any pro guard information it he documentation though.

bpappin, 2015-10-16:
Adding the pro guard config seemed to alleviate the issue. I would recommend that the pro guard configuration be included in the basic documentation.

closed 2015-10-16 by bpappin

commonsguy, 2015-10-16:
Hmmm... I am surprised it had an impact, if you're using a `debug` build. That being said, I am glad the problem is behind you. Issue #112 tracks the documentation request.