commonsguy/cwac-cam2 #114: Move FocusMode Out Of AbstractCameraActivity

commonsguy, 2015-10-16:
For the eventual `cam2-support` work (issue #20), we need to get as much stuff out of the activities and fragments as possible. One item is `FocusMode`, presently in `AbstractCameraActivity`. Move this someplace else (not quite certain where...) that can be central to both implementations.

zokipirlo, 2015-10-20:
I would suggest `CameraSelectionCriteria`. Also for `Facing`.

commonsguy, 2015-10-20:
@zokipirlo `FocusMode` has nothing to do with selecting a camera, so `CameraSelectionCriteria` would not make sense. More likely, it will become just a top-level enum. `Facing` could go on `CameraSelectionCriteria`, since that actually is involved in selecting a camera.

commonsguy, 2016-02-15:
Addressed in v0.5.0.

closed 2016-02-15 by commonsguy