commonsguy/cwac-cam2 #115: Camera preview looks 'zoomed'

brenoptsouza, 2015-10-23:
I noticed it in the Picture Playground app, on HTC M8. When comparing the front camera preview with the default camera app, it looks like its a bit zoomed in by default

commonsguy, 2015-10-23:
Possibly. The issue is maintaining the full-bleed preview. It will be "zoomed" to some extent just by that process. Issue #88 is covering adding a switch to disable the full-bleed preview, so you see exactly what is coming off the camera, albeit without the preview image filling the screen. In addition to this possibly being useful in its own right, I will be using it to determine whether or not the "zooming" effect is just the full-screen bleed or if I am overshooting and stretching the image too far.

commonsguy, 2015-11-01:
Yes, based on `EXTRA_DEBUG_SAVE_PREVIEW_FRAME` on `ClassicCameraEngine`, the live preview is square but zoomed beyond the bounds of the `CameraView` along both axes. It should be zoomed only out beyond one axis, to fill the `CameraView` completely.

commonsguy, 2015-11-08:
I have completely revamped the preview, using [a Google sample app](https://github.com/googlesamples/android-Camera2Basic) as a source of ideas. It works better in general. It is not working well (stretched preview) for the Nexus 10 in landscape; my guess is that the problem is with default-landscape devices. I will be rolling this out with an 0.4.0 in the coming weeks, and will be doing a full round of device tests in the process to determine the scope of the stretching problem.

commonsguy, 2015-11-28:
OK, this isn't perfect, but it seems better. Keep tabs on issue #123, as some devices do not give the proper aspect ratio when using the `camera2` APIs. I am blacklisting them, forcing them back to the original `Camera` API, which works.

commonsguy, 2015-11-28:
Dammit, clicked the wrong button... This will be closed shortly though as part of v0.4.0.

commonsguy, 2015-11-28:
Addressed in v0.4.0.

closed 2015-11-28 by commonsguy