commonsguy/cwac-cam2 #119: [Samsung Galaxy Core Prime] Preview quality is bad

gbero, 2015-11-23:
Hello,

I'm making some tests where I'd like to have the preview only using half of the screen.

I simply did it like : 

<LinearLayout
      android:layout_width="match_parent"
      android:layout_height="match_parent"
      android:orientation="vertical">
  
    <FrameLayout
      android:id="@+id/cwac_cam2_preview_stack"
      android:layout_width="match_parent"
      android:layout_weight="1"
      android:layout_height="0dp">
  
      <com.commonsware.cwac.cam2.CameraView
        android:layout_width="match_parent"
        android:layout_height="match_parent"/>
    </FrameLayout>
  
      <FrameLayout
        android:layout_width="match_parent"
        android:layout_weight="1"
        android:layout_height="0dp"/>
  
    </LinearLayout>

On a Nexus 5 the preview displays correctly. But I tested on 2 Samsung phones and the image gets very blurry. Any idea why ?

commonsguy, 2015-11-23:
> I'm making some tests where I'd like to have the preview only using half of the screen

That's not supported.

> Any idea why?

No. You are welcome to try the library's demo apps on those devices. If you get similar results, follow the [contribution guidelines](https://github.com/commonsguy/cwac-cam2/blob/master/CONTRIBUTING.md) and file a fresh issue with the requested details, so that I can attempt to get my hands on this hardware for testing.

Beyond that, please [use Stack Overflow for questions](https://github.com/commonsguy/cwac-cam2#questions), not the issue tracker, bearing in mind that [I am not supporting anything other than the official APIs at this time](https://github.com/commonsguy/cwac-cam2#basic-usage).

closed 2015-11-23 by commonsguy

Liverm0r, 2015-12-19:
Hey, gbero! I found the same problem. Did you find a solution?

gbero, 2015-12-20:
@Liverm0r unfortunately no. I've decided to only use camera2 for FULL supporting devices, and keep camera1 api for all others. Not ideal though...