commonsguy/cwac-cam2 #126: Camera Flash Setting in CameraActivity.

Ajk2rxKamehameha, 2015-12-03:
I want Flash Setting to change via floating action button in CameraFragment. I have added a floating button but I may not able to change flashMode of cameraController somehow. Can you help me so I can know from where should I proceed. Thanks.

commonsguy, 2015-12-03:
Adding flash control to the FAM is something that I need to do.

While you are welcome to play around with classes like `CameraController`, that is outside the scope of what I support.

jaumecolom, 2015-12-11:
I need the flash floating action button too. I don't know how to add it..

Thank you!

batrand, 2016-04-18:
Any updates on this feature? I've dug around and it seems like you've partially implemented the feature, but canToggleFlashMode() is not used anywhere. Also, it's interesting when ClassicCameraEngine doesn't support dynamic flash mode while CameraTwo does. Is there a reason for this? In cwac-camera we were able to call setFlashMode on the CameraFragment without any problem.

commonsguy, 2016-04-18:
Well, this whole thing is troublesome.

My #1 source of incompatibilities in cwac-camera was in attempting to change `Camera.Parameters` on the fly. Hence, your statement of "we were able to call setFlashMode on the CameraFragment without any problem" is somewhat overstating the case, insofar as changing flash modes requires changing `Camera.Parameters`.

This leaves me with four options:

1. Allow changing flash modes on the fly, but do so by closing the camera, then re-opening the camera with the proper flash mode. This will work, and it should be fairly reliable. However, there should be a delay while the switch is being made, roughly comparable to the delay when switching between the front and back camera.

2. Allow changing flash modes on the fly, and do so without closing the camera, and just pray that the `Camera.Parameters` stuff doesn't break. This does not excite me.

3. Allow changing flash modes on the fly, but only with the `camera2` API. This is the path I had started pursuing, as evidenced by the `canToggleFlashMode()` stuff, before punting on the whole issue for now.

4. Do not allow changing flash modes on the fly, which is where we are today.

batrand, 2016-04-18:
I understand now. So which path will you be pursuing? It sounds like you chose 3. and are simply delaying, but I think 1. is quite reasonable, though I guess you find the delay in the camera closing and switching somehow bothers you as it's intrusive to the user experience? If that's the case, perhaps another method in IntentBuilder can be made? I'm thinking about something like this:

```
switchFlashMethod1() // this is more reliable but will experience a delay
switchFlashMethod3() // this is less intrusive but will not work on classic camera engine
```

My current implementation is to cancel the activity and pass along an extra in the Intent, and get the calling activity to create a new cwac-cam2 instance with the new flash mode. It's totally cumbersome. I'm digging around the library trying to implement exactly what you described in 1. Is there a reason you did not choose that approach?

commonsguy, 2016-04-18:
> So which path will you be pursuing?

I haven't decided. Please understand that work on this library is not my top priority. It's not even in my top ten priorities, at least at the moment.

> I guess you find the delay in the camera closing and switching somehow bothers you as it's intrusive to the user experience?

I am guessing that users will expect something more fluid.

> If that's the case, perhaps another method in IntentBuilder can be made?

No matter what I wind up doing, it will need to be configurable. If I implement only option 1, I would choose to go opt-in, as some developers would prefer no user flash control given the delay. If I implement only option 3, I might elect for opt-out, defaulting to allowing flash control on `camera2` devices.

It's not out of the question that I'll go with what I'll call here "option 1+3", which means that for `camera2` devices we make the flash change dynamically and for `Camera` devices we do the close-and-reopen approach.

batrand, 2016-04-18:
I understand, given the amount of your GitHub repositories lol. Thanks for the update! I'll subscribe to this issue for updates. Meanwhile I'll dig around and see what I can do.

batrand, 2016-04-28:
This is how I have achieved the flash mode toggling. I am using my own extension to `CameraFragment`.
**DISCLAIMER: this is not officially supported by the author of the library.**

Add a new floating action button, and on its OnClick event:

```
getController.stop();
ArrayList<FlashMode> newFlashMode = new ArrayList<>();
newFlashMode.add(FlashMode.ALWAYS); // This is the flash mode you want to change to
getController().getEngine().setPreferredFlashModes(newFlashMode);
getController().start();
```

If you are using your own extension of `CameraFragment `you have to change `setPreferredFlashModes` in `CameraEngine `to public.

I have tested it on 0.5.7 and it's working. Until commonsguy implements his official way to change flash modes, this is working elegantly enough for my use, and hopefully for others as well.

vxhviet, 2016-05-04:
Hi @batrand, this seem like total irrelevant to the topic but could I ask how can you extend `CameraFragment` and set `setPreferredFlashMode` to public? Because there's a lot of private and protected method in this lib, the only possible way is to copy over all the code and kind of butcher it which can get messy really soon.

batrand, 2016-05-04:
That's what I did. I decided to to take that route over working directly with the Android camera API. Though commonsguy doesn't support this, I'd rather reuse his efforts over having to reinvent the wheel (and having to deal with the camera APIs directly). 

That said, the only edits I made into cwac-cam2 source files are making approximately 5 methods public. The rest of the modifications I keep in my own `CameraActivity `and `CameraFragment `classes that `extends `from the library originals. I started doing this around version 0.5.6 of the library and has upgraded through the versions to 0.5.9, and it's not been too bad. I mark all edits to the source with `//TODO`s and Android Studio lists them all in one tab, so they are kept tracked.

But that's only because I needed a full-blown custom-UI camera app (it's released on [Google Play](https://play.google.com/store/apps/details?id=com.officience.freemous), check it out if you are curious!). If you only need the flash toggling, I think you can edit the layout to add one FAB and maybe just one localized code block for the function wiring.

commonsguy, 2016-05-04:
@batrand I think that I have mentioned this in other issues, but I don't have a problem with changing the visibility of methods and other non-functional changes to the library to make your sorts of modifications simpler. I do not want to be responsible for trying to debug or otherwise directly support replacing the library's innards, but visibility changes are trivial. If you like, file an issue asking me to make those methods `public`.

batrand, 2016-05-04:
Hey that would be awesome! Will be filing them right away!

vxhviet, 2016-05-04:
Yeah, I kind of like to implement my own layout for the camera but I have real trouble trying to make `CameraFragment` pointing to a different xml. What I've done so far:

Pointing to `MyVideoRecorderActivity` which is extended from `VideoRecorderActivity`
```java
Intent i = new VideoRecorderActivity.IntentBuilder(HomeActivity.this)
                .quality(AbstractCameraActivity.Quality.HIGH)
                .to(new File(saveFolder, DiskUtils.getVideoName()))
                .updateMediaStore()
                .build();

        i.setClass(this, MyVideoRecorderActivity.class);

        startActivityForResult(i, RequestCode.REQUEST_MY_CAMERA);
```   

Then in `MyVideoRecorderActivity`

```java
public class MyVideoRecorderActivity extends VideoRecorderActivity {

    @Override
    protected MyCameraFragment buildFragment() {
        Log.d("MyVideoRecorderActivity", getOutputUri().toString());

        return(MyCameraFragment.newVideoInstance(getOutputUri(),
                getIntent().getBooleanExtra(EXTRA_UPDATE_MEDIA_STORE, false),
                getIntent().getIntExtra(MediaStore.EXTRA_VIDEO_QUALITY, 1),
                getIntent().getIntExtra(MediaStore.EXTRA_SIZE_LIMIT, 0),
                getIntent().getIntExtra(MediaStore.EXTRA_DURATION_LIMIT, 0)));
    }
}
```
Then in `MyCameraFragment`

```java
public class MyCameraFragment extends CameraFragment{
    @Override
    public View onCreateView(LayoutInflater inflater, ViewGroup container, Bundle savedInstanceState) {
    //trying to inflate it with a different xml but it starts getting complicated
    // mostly because of private methods in CameraController and CameraEngine.
    }
}
```

It can be really simple if I just copy over everything and edit only that one line of code but I prefer to keep the source code untouched in case of future upgrade.

vxhviet, 2016-05-05:
Hi @batrand, I've tried your code snippet for enabling torch flash. The method does get called with a brief pause on the camera preview but there's no flash. I've also had the following in the manifest:

```xml
<uses-permission android:name="android.permission.FLASHLIGHT" />
 <uses-feature
        android:name="android.hardware.camera.flash"
        android:required="false" />
``` 

Am I still missing something obvious?

batrand, 2016-05-05:
I'm not sure if the library itself supports torchlight flash. `FlashMode.ALWAYS` is always turn flash on when taking pictures, not when the camera is just previewing. Have you tried taking picture after calling the method? It should then flash when a photo is captured.

vxhviet, 2016-05-05:
Damn, I must have mistaken it with torchlight flash. Basically I want to enable torch light while recording video in case of lacking light in the environment.

Thank you anyway.

commonsguy, 2016-05-11:
Here are some Googly notes on dynamically changing flash modes with the `camera2` API: http://stackoverflow.com/questions/37144657/switch-flash-in-camera2-api/37149446#37149446