commonsguy/cwac-cam2 #128: Cam2 Picure Playground Preview problem

darekxan, 2015-12-08:
Camera acivity opens and quits immediately.

Device: Samsung i9505 5.0.1
Rom: LRX22C.I9505XXUHOJ2

Steps to reproduce:
Compile latest playground app.
Launch Cam2 Picture Playground
Click "shutter" icon
Activity launches and crashes immediately with:
```
12-08 23:15:00.661 10118-10141/com.commonsware.cwac.cam2.playground:cwac_cam2 E/CameraDevice-0-LE: Surface with size (w=1383, h=1845) and format 0x1 is not valid, size not in valid set: [1440x1080, 1280x720, 1056x864, 960x720, 720x480, 640x480, 320x240, 176x144]
12-08 23:15:00.661 10118-10141/com.commonsware.cwac.cam2.playground:cwac_cam2 W/CameraDevice-JV-0: Stream configuration failed
12-08 23:15:00.661 10118-10141/com.commonsware.cwac.cam2.playground:cwac_cam2 E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
```

Additional findings:
1. Issue can be 'workarounded' by checking "Use front-facing camera", switching to landscape, then tapping 'shutter' icon. Preview starts normally. After that its possible to switch to back camera successfully.

commonsguy, 2015-12-08:
I will see if I can reproduce this on an i9500, otherwise this will have to wait until I can get an i9505 for testing.

You might try "Force classic API" as a more straightforward workaround, to see if that helps.

darekxan, 2015-12-08:
Finding 3: Preview sometimes appears during monkey-testing of 'shutter' button, while most of the time it fails. This leads me into thinking that its a timing or threading issue.

commonsguy, 2015-12-09:
Ah, this is reproducible on the i9500. I'll work on it in the next pulse.

commonsguy, 2015-12-15:
Well, the changes I made in 0.4.2 cleared it up on the i9500. I am marking this as closed for now; if the problem still exists on your test device, let me know and I can reopen it.

closed 2015-12-15 by commonsguy