commonsguy/cwac-cam2 #129: Galaxy S5 (SM-G900F): preview doesn't open

virusman, 2015-12-10:
Galaxy S5 (SM-G900F) running Android 5.0 - camera closes immediately after opening, with the following message:
```
12-10 18:21:07.045 21068-21103/com.commonsware.cwac.cam2.demo:cwac_cam2 E/CameraDevice-0-LE: Surface with size (w=1080, h=1920) and format 0x1 is not valid, size not in valid set: [1920x1080, 1440x1080, 1280x720, 1056x864, 960x720, 800x480, 720x480, 640x480, 352x288, 320x240, 176x144]
12-10 18:21:07.045 21068-21103/com.commonsware.cwac.cam2.demo:cwac_cam2 W/CameraDevice-JV-0: Stream configuration failed
12-10 18:21:07.045 21068-21103/com.commonsware.cwac.cam2.demo:cwac_cam2 E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
```
It only works if the phone is in landscape orientation. Legacy mode doesn't solve the issue.
Looks similar to #128 but in this case the size is correct, just the camera driver doesn't seem to take orientation into account when checking the preview size.

commonsguy, 2015-12-10:
I have an S5 SM-G900H that does not exhibit the problem. So, let's hope that whatever I do with issue #128 fixes it. I am going to put it on the pending-hardware list for now.

virusman, 2015-12-10:
Is it running Android 5.0.0? It may be related to the specific version of Android and not the exact model.

commonsguy, 2015-12-10:
@virusman It is running 5.0.x. Moreover, it wouldn't matter. It's not like I can snap my fingers and get a device to magically sprout another Android OS version. :-)

danwebanddev, 2015-12-14:
Hi,
I have the same issue with a Motorola MotoE2 running Android 5.0.2.
Here is the LogCat:
```
13797-13822/com.commonsware.cwac.cam2.demo:cwac_cam2 I/CameraManager: Using legacy camera HAL.
13797-13822/com.commonsware.cwac.cam2.demo:cwac_cam2 W/ArrayUtils: Ignoring invalid value auto_hdr
13797-13821/com.commonsware.cwac.cam2.demo:cwac_cam2 E/CameraDevice-0-LE: Surface with size (w=637, h=850) and format 0x1 is not valid, size not in valid set: [1280x960, 1280x720, 960x720, 960x540, 864x480, 720x480, 768x432, 640x480, 320x240, 176x144]
13797-13821/com.commonsware.cwac.cam2.demo:cwac_cam2 W/CameraDevice-JV-0: Stream configuration failed
13797-13821/com.commonsware.cwac.cam2.demo:cwac_cam2 E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
```
Please let me know if I should open a new issue.
Thank you!

commonsguy, 2015-12-15:
@virusman @danwebanddev Try 0.4.2 and let me know if it clears up your respective problems.

gbhall, 2016-01-23:
@commonsguy I was wondering if you could tell me how you solved this? Thanks.

commonsguy, 2016-01-23:
@gbhall Well, I don't know that I *did* solve this, since nobody responded. Assuming that it fixed, I moved my `shouldSwapPreviewDimensions` logic into a `OpenedEvent` handler: https://github.com/commonsguy/cwac-cam2/commit/5be7efc164604c5fa2926c697bcd1f371ddce6e3

gbhall, 2016-01-23:
@commonsguy Truth be told I'm also having a go at my own implementation. And only get that error when i switch cameras (rear vs front). What's odd is that even though it complains the size is not in a valid set, if I select a size for the rear, and use that for the front it works (however the ratio of the front will be out). Or if I select a size for the font and use that for the year. It doesn't like changing sizes.  I'm using a single textureview though. Sorry this is probably not the place, however scouring the internet it seems only a few couple of people including you have encountered this.

commonsguy, 2016-02-06:
Closing as assumed fixed, until somebody issues a fresh report.

closed 2016-02-06 by commonsguy