commonsguy/cwac-cam2 #133: LG GPad F (V496) Camera hangs up after opening.

Haris07, 2015-12-17:
Looks like camera is not initializing on this device. It hangs up and remains in this mode.
See attached image
[image]

Also here are some logs related to camera which might help you.
W/CameraDevice-JV-0: Stream configuration failed
E/CameraDevice-0-LE: Surface with size (w=889, h=1186) and format 0x1 is not valid, size not in valid set: [1920x1440, 1920x1080, 1280x960, 1280x800, 1280x720, 1024x768, 960x720, 1024x640, 960x540, 800x600, 864x480, 704x576, 800x500, 720x540, 800x480, 720x480, 640x480, 576x432, 480x320, 384x288, 352x288, 320x240, 240x160, 176x144]
E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed

commonsguy, 2015-12-17:
@Haris07 The error you cited sounds like the one from issue #128, which I addressed in v0.4.2. Make sure that you are testing v0.4.2.

commonsguy, 2016-02-06:
Closing as not reproducible, due to lack of feedback.

closed 2016-02-06 by commonsguy