commonsguy/cwac-cam2 #134: Nexus 5X Compatibility

georgelin422, 2015-12-23:
I tried to run samples on Google Nexus 5X LGE/bullhead Android 6.0 and it seems to fail when sending intent to start Cam2 Activity with rear camera.

error log:
`cwac_cam2 E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed`

commonsguy, 2015-12-23:
Please provide the complete steps to reproduce the crash. Thanks!

hijamoya, 2015-12-24:
Same issue. 
Step:
1. find a  Nexus 5X LGE/bullhead
2. install play ground
3. try to launch the back camera.
4. fail to launch

commonsguy, 2015-12-26:
Neither bug report noted that the actual problem can only be reproduced in portrait mode with the rear-facing camera. Can be seen in `demo/` and `demo-playground/`.

commonsguy, 2015-12-26:
Seems to be an issue of too large of a preview size. I have added some caps that clears up the problem. The bug fix will be released tomorrow, barring unforeseen circumstances.

commonsguy, 2015-12-27:
Addressed in v0.4.3.

closed 2015-12-27 by commonsguy