commonsguy/cwac-cam2 #135: Wrong orientation on android 6 (Marshmallow)

Zuhair-Kj, 2015-12-29:
I am not sure whether this is because of the device (Nexus 5X) or the new OS version
but the camera orientation is reverted see image in link:
https://drive.google.com/file/d/0ByP6surFrliGTUtGaldPcExYQ0E/view?usp=sharing

commonsguy, 2015-12-29:
Please provide complete steps to reproduce the problem.

Zuhair-Kj, 2015-12-30:
@commonsguy 
```java
intent = new CameraActivity.IntentBuilder(QuickSellActivity.this).skipConfirm()
                            .to(tempOutputFileUri)
                            .forceClassic()
                            .build();
startActivityForResult(intent, IMAGE_CAPTURE_CODE);
```

commonsguy, 2015-12-30:
Before filing other issues, please read the contribution guidelines for [instructions on submitting bug reports](https://github.com/commonsguy/cwac-cam2/blob/master/CONTRIBUTING.md#contributing-bug-reports).

I can reproduce your problem using the `demo-playground` app, with "Save full image to file" checked, "Force classic API" checked, and everything else at their defaults, from a fresh install of that app.

commonsguy, 2016-01-03:
Duplicate of issue #130

closed 2016-01-03 by commonsguy

commonsguy, 2016-01-03:
Try v0.4.4, as I have added the Nexus 5X to the alternative orientation algorithm whitelist, and that seems to clear up the problem.