commonsguy/cwac-cam2 #141: Appears to crash immediately on the LG V10 H901

patrickms, 2016-01-30:
I tried an app using cwac-cam2 on an LG V10 H901 and it crashed immediately. Unfortunately I didn't have crash logging set up and the phone isn't mine, so i don't have any more information. If someone who has access to an LG V10 H901 (http://www.lg.com/us/cell-phones/lg-H901-Space-Black) could try it, that would be very helpful.

commonsguy, 2016-05-14:
Well, I couldn't get either the `demo` or the `demo-playground` app to crash, though the `demo` app was not liking the rear-facing camera much. I moved this device to use the `ClassicCameraEngine`, which cleared up that issue. Unless somebody can give me a more concrete scenario, I will assume that this, or past bug fixes, has cleared up this problem.

commonsguy, 2016-05-15:
Addressed in v0.5.11.

closed 2016-05-15 by commonsguy