commonsguy/cwac-cam2 #143: Galaxy S6

julioyg, 2016-02-03:
I got an issue trying to take a picture with a galaxy S6: press the capture button, it turns grey but the photo is not being taken till the focus changes dramatically like covering the camera. it works if I add that device as problematic `isProblematicDeviceOnNewCameraApi()`

commonsguy, 2016-02-03:
Can you provide me with the specific `Build.PRODUCT` and `Build.MANUFACTURER` values for this device? The latter is probably `samsung`. Thanks!

julioyg, 2016-02-09:
sorry about the time I took to answer. I added these conditions to you method `isProblematicDeviceOnNewCameraApi()`:

```
    if ("samsung".equals(Build.MANUFACTURER) &&
            "zerofltexx".equals(Build.PRODUCT)) {
      return(true);
    }
```

that's what I got from the device

julioyg, 2016-02-09:
wrong button

commonsguy, 2016-02-09:
OK, that's the SM-G920F. I'll see if I can get my hands on one someday.

In the meantime, I'll look to add your values to `isProblematicDeviceOnNewCameraApi()` for the next update.

Thanks!

commonsguy, 2016-02-15:
Addressed in v0.5.0.

closed 2016-02-15 by commonsguy