commonsguy/cwac-cam2 #148: Unable to take Photos due to generic camera2 capture failure

aftabsikander, 2016-02-11:
Hi,
During our regression testing for our Product, some user faced a issue they were unable to take photos within our application which was using your library.Below is the Build.MANUFACTURER and Build.PRODUCT along with hardware name. When they click capture button nothing happens and exception is throw in our user session. We were using **'com.commonsware.cwac:cam2:0.4.2'** now I am upgrading to your latest version. In the mean while Below find the detail important below.

**Manufacturer: Samsung
Product:  jfltespr
Device Model: SPH-L720**

Below is what I am using for CameraActivity.IntentBuilder configuration

`Uri makePhotoUri = Uri.fromFile(fileName);`
`mCapturedImageUri = makePhotoUri;`
`Intent cameraIntent = new CameraActivity.IntentBuilder(getActivity())
  .facing(CameraActivity.Facing.BACK)
  .debug()
  .to(mCapturedImageUri)
  .build();`

The exception what I am getting from my user recorded sessions.

`java.lang.RuntimeException: generic camera2 capture failure at com.commonsware.cwac.cam2.CameraTwoEngine$CapturePictureTransaction.onCaptureFailed(SourceFile:539)`
	`at java.lang.reflect.Method.invoke(Native Method)`
	`at java.lang.reflect.Method.invoke(Method.java:372)`
	`at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)`
	`at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)`
	`at android.os.Handler.handleCallback(Handler.java:739)`
	`at android.os.Handler.dispatchMessage(Handler.java:95)`
	`at android.os.Looper.loop(Looper.java:145)`
	`at android.os.HandlerThread.run(HandlerThread.java:61)`

Let me know what I am doing wrong, Or you need more information about this issue let me know.

commonsguy, 2016-02-11:
There is not much that I can suggest. I will work on acquiring one of these devices, and I will see if I can reproduce the problem.

You might wish to switch to the classic camera API (`forceClassic()` on the `IntentBuilder`), at least for this `Build.MANUFACTURER`/`Build.PRODUCT` combination.

aftabsikander, 2016-02-11:
What If I use `ForceClassic()` method for overall `Build.MANUFACTURER/Build.PRODUCT` will it temporary solve this issue until we find a proper solution. I can't hard code `Build.MANUFACTURER/Build.PRODUCT` as there are so many device which we are targeting.

commonsguy, 2016-02-11:
> will it temporary solve this issue until we find a proper solution

I have no idea, as I do not know what the issue is. Your stack trace is tied to the `camera2` API, and so `forceClassic()` may help, or it may not help, or it may make things worse for this device. I have no way to know.

aftabsikander, 2016-02-11:
I am using `ForceClassic()` for now, will let you know if any issue still comes after this update. In the meanwhile can you find out the cause of this issue

commonsguy, 2016-02-11:
> In the meanwhile can you find out the cause of this issue

As I wrote [previously](https://github.com/commonsguy/cwac-cam2/issues/148#issuecomment-182936187), I will work on acquiring one of these devices, and I will see if I can reproduce the problem.

commonsguy, 2016-02-18:
I cannot reproduce your problem using the `demo-playground` app, using the rear-facing camera, with debug mode on, and asking it to store the image to a `Uri`.

I *am* having problems with the main `demo` app with the rear-facing camera, and it is possible that when I clear that up, it will clear up your issue here. However, I have no way to confirm that.

commonsguy, 2016-02-18:
OK, the problem with the `demo` app is related to flash modes. I have logged issue #151 for that. However, that makes it rather unlikely that the `demo` app problem is the same problem that you are having.

In order to proceed on your issue, I will need you to provide me with a project that can reproduce the problem on your device.

aftabsikander, 2016-02-23:
Really not sure what is the issue which is causing this problem, I have created a demo project for you to debug or try to reproduce this issue. The demo project is working fine on my device. This issue is seen on most of the **Beta Testers devices** during our application **UAT phase**

This project contains same structure like in our live application may be this will help you find the issue. 

**Link to Demo Project**
[https://drive.google.com/file/d/0BzFEs0fNIo-DWU0tTmd6U051dFU/view?usp=sharing](url)

**Link to Demo Project APK File**
[https://drive.google.com/file/d/0BzFEs0fNIo-DaW0yYkdqWlo4cUk/view?usp=sharing](url)

commonsguy, 2016-02-27:
I downloaded your source code. The only change that I made was to drop back to version 1.5.0 of the Android Plugin for Gradle and build with Gradle 2.5. Given that, I cannot reproduce your problem by running the app and taking a couple of pictures on a SPH-L720.

What are the steps that I am supposed to do that are supposed to reproduce this problem?

aftabsikander, 2016-03-01:
Please find the **GIF image** which highlights the steps taken from `reported tester recorded session`. The steps are simple they click capture button after waiting for 3 seconds a white background screen comes up with close `'X' Mark on Action Bar`

[image]

commonsguy, 2016-03-01:
I definitely took pictures with the SPH-L720 in portrait mode with the rear-facing camera. I did not encounter this problem with your sample app.

aftabsikander, 2016-03-01:
The above steps were taken from various recording session of same user. He faced this issue with the above steps many times. This issue may have existed in your previous version, As I was using `com.commonsware.cwac:cam2:0.4.2` and the sample which I shared is based on ` com.commonsware.cwac:cam2:0.5.+`. Given that I believe the issue may have been with `V 4.2`.

I have shared more detail about the device below.

**Manufacturer: Samsung**
**Product: jfltespr**
**Device Model: SPH-L720**
**osRelease: 5.0.1**
**osVersion: 3.4.0-4745659**

Do let me know if you need any more information.

commonsguy, 2016-03-01:
If you are able to reproduce this sometime with a 0.5.x edition of the library, let me know and I can reopen the issue. Otherwise, I'll hope that I fixed it. :-)

Thanks for your help!

closed 2016-03-01 by commonsguy