commonsguy/cwac-cam2 #160: Bug on open front camera: previewSize must not be wider than activeArray

Liverm0r, 2016-03-01:
Highscreen Pure F, android 5.0.2
http://joxi.ru/a2XVo3OSyV8K3r

commonsguy, 2016-03-01:
It is not possible for me to get a Highscreen Pure F here in the US, so it is not likely that I will be able to address this.

For future reference, what did you do, with the `demo/` or `demo-playground/` apps, that triggered this crash?

Liverm0r, 2016-03-01:
Bug happens immediately when onCreate is triggered. I tried the same code on htc one mini 2, it's alright.

`
public class CwacActivity extends AppCompatActivity {

    private static final int REQUEST_PORTRAIT_FFC = 1271;
    private String mPath;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_cwac);

        mPath = Environment.getExternalStorageDirectory() + "/portrait-front.jpg";

        Intent i = new CameraActivity.IntentBuilder(CwacActivity.this)
                .skipConfirm()
                .facing(Facing.FRONT)
                .to(new File(mPath))
                .zoomStyle(ZoomStyle.SEEKBAR)
                .debug()
                .updateMediaStore()
                .build();

        startActivityForResult(i, REQUEST_PORTRAIT_FFC);
    }

    @Override
    protected void onActivityResult(int requestCode, int resultCode, Intent data) {
        if (requestCode == REQUEST_PORTRAIT_FFC) {

            Intent intent = new Intent(CwacActivity.this, MainActivity.class);
            intent.putExtra("path", mPath);
            startActivity(intent);
        }
    }
}
`

commonsguy, 2016-03-01:
Well, if I happen to somehow acquire one of those devices, I can try to research this.

From the stack trace, it feels like it is tied to zoom. You might try removing `zoomStyle(ZoomStyle.SEEKBAR)` and see if you have better luck. If you do, you could conditionally use `zoomStyle(ZoomStyle.SEEKBAR)`, skipping it on this device. The `Build.MANUFACTURER` and `Build.PRODUCT` fields are what I use for this sort of "route around the damage" per-device behavior.

Liverm0r, 2016-03-01:
I tried to remove zoomStyle(ZoomStyle.SEEKBAR). Didn't help.

> one of those devices

I meant that the code work fine with htc one mini 2

commonsguy, 2016-03-01:
And by "one of those devices", I meant the Highscreen Pure F. I apologize for any confusion.