commonsguy/cwac-cam2 #165: Unable to build release APK when using ProGuard

immahacker, 2016-03-06:
i have tried putting this line to work:
-keep class com.commonsware.cwac.** { *; }

came out with this error:

Information:Gradle tasks [:app:assembleRelease]
:app:preBuild UP-TO-DATE
:app:preReleaseBuild UP-TO-DATE
:app:checkReleaseManifest
:app:preDebugBuild UP-TO-DATE
:facebook:compileLint
:facebook:copyReleaseLint UP-TO-DATE
:facebook:preBuild UP-TO-DATE
:facebook:preReleaseBuild UP-TO-DATE
:facebook:checkReleaseManifest
:facebook:preDebugAndroidTestBuild UP-TO-DATE
:facebook:preDebugBuild UP-TO-DATE
:facebook:prepareComAndroidSupportSupportV42221Library UP-TO-DATE
:facebook:prepareReleaseDependencies
:facebook:compileReleaseAidl UP-TO-DATE
:facebook:compileReleaseRenderscript UP-TO-DATE
:facebook:generateReleaseBuildConfig UP-TO-DATE
:facebook:generateReleaseAssets UP-TO-DATE
:facebook:mergeReleaseAssets UP-TO-DATE
:facebook:generateReleaseResValues UP-TO-DATE
:facebook:generateReleaseResources UP-TO-DATE
:facebook:mergeReleaseResources UP-TO-DATE
:facebook:processReleaseManifest UP-TO-DATE
:facebook:processReleaseResources UP-TO-DATE
:facebook:generateReleaseSources UP-TO-DATE
:facebook:compileReleaseJava UP-TO-DATE
:facebook:extractReleaseAnnotations UP-TO-DATE
:facebook:mergeReleaseProguardFiles UP-TO-DATE
:facebook:processReleaseJavaRes UP-TO-DATE
:facebook:packageReleaseJar UP-TO-DATE
:facebook:compileReleaseNdk UP-TO-DATE
:facebook:packageReleaseJniLibs UP-TO-DATE
:facebook:packageReleaseLocalJar UP-TO-DATE
:facebook:packageReleaseRenderscript UP-TO-DATE
:facebook:packageReleaseResources UP-TO-DATE
:facebook:bundleRelease UP-TO-DATE
:layer-atlas:compileLint
:layer-atlas:copyReleaseLint UP-TO-DATE
:layer-atlas:preBuild UP-TO-DATE
:layer-atlas:preReleaseBuild UP-TO-DATE
:layer-atlas:checkReleaseManifest
:layer-atlas:preDebugAndroidTestBuild UP-TO-DATE
:layer-atlas:preDebugBuild UP-TO-DATE
:layer-atlas:prepareComAndroidSupportAppcompatV72221Library UP-TO-DATE
:layer-atlas:prepareComAndroidSupportDesign2221Library UP-TO-DATE
:layer-atlas:prepareComAndroidSupportMediarouterV72220Library UP-TO-DATE
:layer-atlas:prepareComAndroidSupportRecyclerviewV72221Library UP-TO-DATE
:layer-atlas:prepareComAndroidSupportSupportV42221Library UP-TO-DATE
:layer-atlas:prepareComDavemorrisseyLabsSubsamplingScaleImageView341Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServices830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesAds830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesAnalytics830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesAppindexing830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesAppinvite830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesAppstate830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesAuth830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesBase830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesBasement830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesCast830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesDrive830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesFitness830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesGames830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesGcm830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesIdentity830Library UP-TO-DATE
:layer-atlas:prepareComGoogleAndroidGmsPlayServicesLocation830Library UP-TO-DATE
[... 87 lines cut ...]
:app:lintVitalRelease
:app:compileReleaseNdk UP-TO-DATE
:app:collectReleaseMultiDexComponents UP-TO-DATE
:app:proguardRelease
Note: there were 364 duplicate class definitions.
      (http://proguard.sourceforge.net/manual/troubleshooting.html#duplicateclass)
Warning:com.commonsware.cwac.cam2.AbstractCameraActivity: can't find referenced method 'void requestPermissions(java.lang.String[],int)' in program class com.commonsware.cwac.cam2.AbstractCameraActivity
Warning:com.commonsware.cwac.cam2.AbstractCameraActivity: can't find referenced method 'int checkSelfPermission(java.lang.String)' in program class com.commonsware.cwac.cam2.AbstractCameraActivity
Warning:there were 2 unresolved references to program class members.
         Your input classes appear to be inconsistent.
         You may need to recompile the code.
         (http://proguard.sourceforge.net/manual/troubleshooting.html#unresolvedprogramclassmember)
:app:proguardRelease FAILED
Error:Execution failed for task ':app:proguardRelease'.
> java.io.IOException: Please correct the above warnings first.
Information:BUILD FAILED
Information:Total time: 1 mins 7.069 secs
Information:1 error
Information:3 warnings
Information:See complete output in console

solved the crash by adding the line:
-dontwarn com.commonsware.cwac.**

but .APK size was severely reduced into half (normal was 10MB, now 5MB), rendering the app unresponsive.

commonsguy, 2016-03-06:
`requestPermissions()` is inherited from `Activity`, and `checkSelfPermission()` is inherited from `Context`, using `compileSdkVersion 23`. All uses of those methods are wrapped in appropriate `Build` tests so that they are called only on API Level 23+ devices.

Both of the demo projects are set with `compileSdkVersion 23`. Your error is not coming from either demo project, as neither demo project uses Play Services.

You may wish to try building one of the demo projects, using your desired ProGuard settings, and see what happens. If that fails, attach your ProGuard configuration file to a comment here along with other details (e.g., which demo project you tried), and I will see if I can reproduce the problem.

commonsguy, 2016-03-26:
Closing as not reproducible.

closed 2016-03-26 by commonsguy