commonsguy/cwac-cam2 #166: Take multiple Pictures

cschnelting, 2016-03-07:
hi, i would like to take multiple pictures with your CameraActivity. At the moment i Build an Intent with CameraActivity.IntentBuilder, start the Intent with startActivityForResult and get the Picture in onActivityResult. But i can only take one Picture on every startActivityForResult().

Maybe it's possible to set an int "maxPictures" on CameraActivity.IntentBuilder and receive all Pictures as String-Array in onActivityResult? The CameraActivity should return all Picture-Paths if the "maxPictures"-Value is reached or the User has pressed the return-Button.

Are you interested in this feature? And if you're interested, how would you implement this feature?

commonsguy, 2016-03-07:
> Are you interested in this feature?

It's not an itch that I need to scratch at the present time, if that is what you mean.

> how would you implement this feature?

Off the cuff, it might require a substantial rewrite of the library. Your proposed API seems reasonable, but there is a fair bit of code that assumes that we are only taking one picture at a time.

cschnelting, 2016-03-10:
Ok, we will solve it ourselves. Thank you!

closed 2016-03-10 by cschnelting