commonsguy/cwac-cam2 #167: Crash on rotate camera twice

VictorHVS, 2016-03-14:
`03-14 10:13:55.404 21426-21426/br.net.infatec.diariosincronizado E/Event: Could not dispatch event: class com.commonsware.cwac.cam2.CameraEngine$OrientationChangedEvent to subscribing class class com.commonsware.cwac.cam2.CameraController
                                                                          java.lang.RuntimeException: Method called after release()
                                                                              at android.hardware.Camera.setDisplayOrientation(Native Method)
                                                                              at com.commonsware.cwac.cam2.plugin.OrientationPlugin$Classic.configureStillCamera(OrientationPlugin.java:152)
                                                                              at com.commonsware.cwac.cam2.ClassicCameraEngine$Session.configureStillCamera(ClassicCameraEngine.java:559)
                                                                              at com.commonsware.cwac.cam2.ClassicCameraEngine.handleOrientationChange(ClassicCameraEngine.java:350)
                                                                              at com.commonsware.cwac.cam2.CameraController.onEventMainThread(CameraController.java:362)
                                                                              at java.lang.reflect.Method.invokeNative(Native Method)
                                                                              at java.lang.reflect.Method.invoke(Method.java:515)
                                                                              at de.greenrobot.event.EventBus.invokeSubscriber(EventBus.java:498)
                                                                              at de.greenrobot.event.EventBus.postToSubscription(EventBus.java:433)
                                                                              at de.greenrobot.event.EventBus.postSingleEventForEventType(EventBus.java:410)
                                                                              at de.greenrobot.event.EventBus.postSingleEvent(EventBus.java:383)
                                                                              at de.greenrobot.event.EventBus.post(EventBus.java:263)
                                                                              at com.commonsware.cwac.cam2.plugin.OrientationPlugin$1.onOrientationChanged(OrientationPlugin.java:60)
                                                                              at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
                                                                              at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:418)
                                                                              at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                              at android.os.MessageQueue.next(MessageQueue.java:138)
                                                                              at android.os.Looper.loop(Looper.java:123)
                                                                              at android.app.ActivityThread.main(ActivityThread.java:5214)
                                                                              at java.lang.reflect.Method.invokeNative(Native Method)
                                                                              at java.lang.reflect.Method.invoke(Method.java:515)
                                                                              at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:814)
                                                                              at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:630)
                                                                              at dalvik.system.NativeStart.main(Native Method)`

After that...

`03-14 10:13:55.504 21426-23108/br.net.infatec.diariosincronizado E/CWAC-Cam2: Exception in camera processing
                                                                              java.io.IOException: setPreviewTexture failed
                                                                                  at android.hardware.Camera.setPreviewTexture(Native Method)
                                                                                  at com.commonsware.cwac.cam2.ClassicCameraEngine$3.run(ClassicCameraEngine.java:254)
                                                                                  at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                                  at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                                  at java.lang.Thread.run(Thread.java:841)`

commonsguy, 2016-03-14:
@VictorHVS Are you able to reproduce this with one of the sample apps? If so, which?

Also, what device(s) are giving you these results?

Thanks!

VictorHVS, 2016-03-14:
Yes, "demo" sample. Device: Sony / D2104 (Xperia E1)

commonsguy, 2016-04-02:
I doubt that I will ever be able to get my hands on this device, so hopefully this will get cleared up by some other means.