commonsguy/cwac-cam2 #168: Incorrect ratio with CamcorderProfile.QUALITY_HIGH

ngochuu90, 2016-03-15:
phone: LG G3, android 5.0.2

commonsguy, 2016-03-15:
Please follow the instructions for [how to file a bug report](https://github.com/commonsguy/cwac-cam2/blob/master/.github/CONTRIBUTING.md#contributing-bug-reports).

ngochuu90, 2016-03-15:
This is my code:

File file = new File(this.getExternalFilesDir(null), "test.mp4");
Intent result = new VideoRecorderActivity.IntentBuilder(this)
                .to(file)
                .quality(VideoRecorderActivity.Quality.HIGH)
                .durationLimit(5000)
                .focusMode(FocusMode.CONTINUOUS)
                .build();

Video incorrect ratio
[image]

Image correct ratio
[image]

Thanks!

commonsguy, 2016-03-15:
@ngochuu90 OK. In order for me to try to reproduce this, I will need the `Build.MANUFACTURER` and `Build.PRODUCT` values for your device, as I may need to go buy one. The `MANUFACTURER` should be `LG` (based on my other LG device). There may be several models that are marketed as the "G3", which is why I need the specific `Build.PRODUCT` value.

commonsguy, 2016-03-26:
@ngochuu90 I repeat: I need the `Build.MANUFACTURER` and `Build.PRODUCT` values for your device.

ngochuu90, 2016-03-29:
@commonsguy 
My phone information
 - Build.MANUFACTURER: LGE LG-D851
 - Build.PRODUCT            : g3_tmo_us

In addition, the library can not be used in the Android version 4.0: Camera Preview freezes after click to record. And click stop, app is forced stop

commonsguy, 2016-04-02:
The Cam2 code does not set a resolution for videos. It uses `CamcorderProfile` and `setProfile()` on `MediaRecorder` for an available profile. For this device, `CamcorderProfile.QUALITY_HIGH` is giving the erroneous output, and `CamcorderProfile.QUALITY_LOW` is giving low-quality output but with the proper aspect ratio.

I will experiment with other `CamcorderProfile` values for this device to see if I can work out a substitute that provides the proper aspect ratio and better-than-`LOW` quality.

commonsguy, 2016-04-03:
Well, `QUALITY_2160P` is not supported, and both `QUALITY_1080P` and `QUALITY_720P` show the aspect ratio problem. I am settling for `QUALITY_480P`, which seems better.

commonsguy, 2016-04-03:
Addressed in v0.5.5.

closed 2016-04-03 by commonsguy