commonsguy/cwac-cam2 #175: Capturing picture does not return to calling activity

batrand, 2016-04-08:
Hello,

I'm trying to use the library, and am following the documentation. I got the activity to open, but after capturing the photo, the preview freezes for a few seconds, the capture button greys out, but instead of returning to MainActivity, the preview restarts instead. I put debug output in onActivityResult and nothing fired, and no picture shows up in the file browser either.

Here is how the intent is built:

```
CameraActivity.IntentBuilder b = new CameraActivity.IntentBuilder(MainActivity.this);
b.to(new File(appPicDir,"cwactest.jpg")) 
                .zoomStyle(ZoomStyle.SEEKBAR)
                .focusMode(FocusMode.CONTINUOUS)
                .allowSwitchFlashMode()
                .updateMediaStore()
                .facing(Facing.BACK);
```

Then I started it:
`startActivityForResult(b.build(),7896);`

Apparently I did something wrong, but can't seem to figure out where.

batrand, 2016-04-08:
I have created a new project and tried again. The issue remains. After pressing the capture button, preview freezes for a few seconds then restarts, the capture button turns from green to gray. It does not return to MainActivity.

I pasted the code here: [gist](https://gist.github.com/batrand/af0aa310210b47a08f20219ab2fe4747).

batrand, 2016-04-08:
The issue seems to be device specific (OnePlus X) and only to the back camera as I tested it again on a Nexus 5 and it works correctly. 

I tried the demo and it's the same"preview freezes for a few seconds then resume, capture button grays out and it stays like that: preview running forever). After a while (literally minutes, not seconds) I then tried pressing back, which shows the welcome screen with both previous and capture button grays out. It stays like that for a while and then it automatically opens the front-facing camera. I captured that and it proceeds as intended.

**Summary**: OnePlus X is problematic. Front camera works correctly, Back camera does not resume to calling activity after capture finishes but stays on the preview indefinitely with capture and switch buttons disabled.

I uploaded the zip file [here](https://goo.gl/Ie3a8P). Hope this helps. I can provide more information if needed.

batrand, 2016-04-08:
Update: I added forceClassic() and it works perfectly. The demo-playground has the same issue with the back camera, and is also resolved by calling forceClassic()

commonsguy, 2016-04-08:
I need the `Build.MANUFACTURER` and `Build.PRODUCT` values for the hardware on which you are experiencing the problem. For all I know, there are several devices marketed as the "OnePlus X", just as there are several Samsung Galaxy S6's.

batrand, 2016-04-09:
`Build.MANUFACTURER` = `OnePlus`
`Build.PRODUCT` = `OnePlus`

[image]

That doesn't look very helpful... the phone's About screen reports its "Model number" as "ONE E1001", if that helps.

commonsguy, 2016-04-09:
OK, thanks! That's the Chinese variant (versus the E1003 for the rest of Asia plus Europe and E1005 for North America). It does not appear likely that I will be able to get an E1001. However, I do not have any OnePlus devices in my lab, so I will try to get *some* OnePlus X, and I'll see if I can reproduce the problem.

commonsguy, 2016-04-14:
OK, I can reproduce the problem on an E1005, but only in portrait mode. Two questions:

1. Is that what you are seeing as well, that landscape is fine but portrait freezes?

2. Is `Build.MODEL` for your device `ONE E1001`? Mine is `ONE E1005`, so I am hoping they were consistent across the models (even if they do not use `Build.PRODUCT` like other manufacturers seem to).

I will configure the library to force this device into classic mode by default.

Thanks!

batrand, 2016-04-15:
1. I can confirm that it works fine in landscape mode but freezes in portrait.
2. Yes, Its `Build.Model` returns `ONE E1001`

Many thanks for the prompt resolution, by the way!

commonsguy, 2016-04-17:
Addressed in 0.5.7.

closed 2016-04-17 by commonsguy

protocol10, 2017-03-16:
Is this fix available, I was able to produce on the following device, is using forceClassic only way to fix it. If we apply this as a fix will it not affect for any other manufacturers

[image]

commonsguy, 2017-03-16:
@protocol10 As was noted above, this was fixed, but for the `E1001`. Please provide me with the `Build.PRODUCT` value for your device, to confirm that yours is an `E1003`.