commonsguy/cwac-cam2 #178: Preview Image is always in Bad quality

aftabsikander, 2016-04-14:
I have been observing this Bad quality preview issue for a while, I have tried you `demo` and `Playground` sample and it can be easily be reproduced or observed when using following config. I am using `0.5.6 version.`
```
 Intent cameraIntent = new CameraActivity.IntentBuilder(getActivity())
                .facing(Facing.BACK)
                .debug()
                .facingExactMatch()
                .quality(AbstractCameraActivity.Quality.HIGH)
                .forceClassic()
                .build();
```
List of Devices which were used to produce this issue

**Device # 1**
```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```
**Device # 2**

```
Brand: samsung
DeviceModel: SM-G530H
Product: fortuna3gxx
Manufacturer: samsung
Android Version: 4.4.4
```

**Device # 3**

```
Brand: samsung
DeviceModel: SM-G530H
Product: fortuna3gxx
Manufacturer: samsung
Android Version: 5.0
```
**Note**
Image quality is great after the `skip screen`, Only during Confirmation screen Image is shown in bad quality. One Option for me is to use `.skipConfirm()` for my intent builder but this is not prefered solution for my users.

commonsguy, 2016-04-14:
I just tried your configuration on a Nexus 5 running 6.0.1, using the `demo-playground` app. I do not consider the confirmation screen edition of the image to be "bad quality".

Since "bad quality" is a subjective measure, you are going to need to somehow convince me that:

- The confirmation screen image has "bad quality", and
- That this is something that I can somehow fix

So, let's start with the first one. What is your evidence that the confirmation screen image has "bad quality"?

aftabsikander, 2016-04-14:
See the attached photos, `Picture-1` is the frame where user clicked capture button, assuming this frame would be the final output for the confirmation screen edition.
Now take a look at `Picture-2` this is the final output after clicking capture button.
[image]
[image]

What user is seeing the picture is blurred and unreadable.

commonsguy, 2016-04-14:
Part of what you are seeing is that you are testing on some device that does not save portrait images as portrait, but rather saves them as landscape with the EXIF `Orientation` header to tell image viewers to rotate the output. You should have better results if you take the picture in landscape on that device. Dealing with this EXIF header is mostly issue #15, though I just filed issue #179 for rotating the preview.

Beyond that, I am intentionally limiting the resolution of the confirmation image to ~2MB of heap space, given Android's heap limitations, to try to strike a balance between preview quality and crashing with an `OutOfMemoryError`.

So, take the same device that you used for your previous comment, and try taking a picture of the same document, but this time in landscape. I do not have your document, but in tests that I just ran on a Nexus 5, written text is definitely fuzzy due to the downsampling of the preview image, but it is nowhere near as bad as what your screenshots show. If the landscape edition of your document photo preview is acceptable, then perhaps issue #179 will be sufficient.

aftabsikander, 2016-04-14:
I just ran a sample Test of some document on my Nexus 5 I still believe preview image should be in same quality as during capture framed. downsampling of preview image may lead to unknown cases where user would feel picture is not coming as per device own camera app.

see attached images
[image]
[image]

commonsguy, 2016-04-14:
> I still believe preview image should be in same quality as during capture framed

They will not necessarily be "in same quality" no matter what I do. The Android camera code that is delivering preview frames to the `TextureView` has very little to do with what I am doing for the confirmation screen. The scaling is done in hardware, the image formats are not the same, and so on.

I have filed issue #180 to allow for greater configurability of the confirmation image quality. You can then set that to be as high as you like, and if it wrecks your app's heap, lower the quality factor.

closed 2016-04-14 by commonsguy