commonsguy/cwac-cam2 #180: Allow Configuration of Confirmation Preview Quality

commonsguy, 2016-04-14:
Expose `EXTRA_CONFIRMATION_QUALITY` and an equivalent `CameraActivity.IntentBuilder` method. This will be a `float` from 0.0f to 1.0f, representing the fraction of the heap limit to attempt to use for the confirmation preview image. Calculating this would take the place of the hardcoded 2000000 value used presently.

Once the stub implementation for this is in place, try 1.0f and see what the result is that we get from `BitmapFactory.decodeByteArray()` (`null`? `OutOfMemoryError`? something else?). When we get that condition, halve the quality fraction and try again, until we get success or the fraction falls below some threshold (0.001f or so), at which point we give up.

Set the default to be something fairly low (probably 0.1f or 0.05f) to simulate the current value.

Use `PackageManager` to determine if the app has requested a large heap, and if so, calculate the fraction off of the large-heap limit, not the regular heap limit.

commonsguy, 2016-04-14:
Temporarily, if the extra is not provided, use the existing hard-coded value (2000000). This will allow this enhancement to be added without affecting existing devices. Then, on the next minor release, cut over to using the extra or a default heap-size fraction all of the time.

aftabsikander, 2016-04-15:
When this feature would be included in next release any idea ?

commonsguy, 2016-04-15:
Possibly as early as this weekend. That being said, please do not count on any particular release schedule for any of the issues in this issue tracker.

aftabsikander, 2016-04-15:
Meanwhile I would be creating custom Preview screen within my application until this is fix in next schedule release.

commonsguy, 2016-04-15:
The OOM facet of this is moved into issue #183, since that's really a separate problem and needs to handle the thumbnail response anyway.

commonsguy, 2016-04-17:
Addressed in 0.5.7.

closed 2016-04-17 by commonsguy