commonsguy/cwac-cam2 #181: unable to use Front Camera on MultiDex enabled Application

aftabsikander, 2016-04-15:
I am unable to open `Front camera` now within my application as recently my application method counts are increased and I had to implement `MultiDex feature`. Previously it was working fine when multiDex was not enabled.  I am using following builder config. 

```
 Intent cameraIntent = new CameraActivity.IntentBuilder(getActivity())
                .facing(Facing.BACK)
                .debug()
                .facingExactMatch()
                .quality(AbstractCameraActivity.Quality.HIGH)
                .forceClassic()
                .build();
```

When I try to open front camera by option,an exception is thrown any solution or workaround to fix this issue. This happens only for Multi Dex Applications

```
com.test.android.test:cwac_cam2 W/System: ClassLoader referenced unknown path: /data/app/com.test.android.test-1/lib/arm
com.test.android.test:cwac_cam2 I/MultiDex: VM with version 2.1.0 has multidex support
com.test.android.test:cwac_cam2 I/MultiDex: install
com.test.android.test:cwac_cam2 I/MultiDex: VM has multidex support, MultiDex support library is disabled.
com.test.android.test:cwac_cam2 I/test: Application object created
com.test.android.test:cwac_cam2 I/test: Application object recreated
```

**Device Information**
```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```

commonsguy, 2016-04-15:
If you publish the source code to a complete sample project that can reproduce your problem, I can try to take a look at it. Comment back here with a link to the source code, and I can reoopen the issue. Otherwise, I have no way to reproduce your situation and cannot do anything.

closed 2016-04-15 by commonsguy

aftabsikander, 2016-04-15:
I cant share the actual project source code what I can do is I will create a sample project which will reproduce this issue for you. I will include lots of libraries and will share a release Apk file as well without pro guard enabled so that its easy to reproduce this MultiDex issue 

If this is alight for you let me know I l start work on this sample

commonsguy, 2016-04-15:
Do not bother with the release APK. However, I will need instructions for how to build an APK from your source that should reproduce the problem, along with instructions for what I need to do in the app that causes the crash (e.g., is it just launching `CameraActivity`, or do I have to do something in there?).

aftabsikander, 2016-04-15:
Alight got your point, Will share source code with you soon Thanks

aftabsikander, 2016-04-15:
I am unable to reproduce that exception in sample project. But what I can reproduce is that `I am unable to open FFC using above configuration`. I have attached the source code.

[Source Code](https://drive.google.com/file/d/0B1fc9uDBXpvVc1Q4eXVULUh1SFU/view?usp=drive_web)

**Screenshot to show How to open Front camera**
[image]

commonsguy, 2016-04-15:
I have filed issue #182 to track this. That FAM should be disabled with your setup, since you are requesting that we only use the rear-facing camera (`facing(Facing.BACK)` plus `.facingExactMatch()`).

aftabsikander, 2016-04-15:
So basically removing this method `.facingExactMatch()` I would be able to use `FFC Feature` ?

commonsguy, 2016-04-15:
Yes. If *that* does not work, there is a separate bug.

aftabsikander, 2016-04-15:
After removing that method its started working, Now I can open `FFC & RFC`