commonsguy/cwac-cam2 #184: CameraTwoEngine NPE in Android N

simonnorberg, 2016-04-15:
I get this exception when I try to capture a picture with the demo apps.

Device: Nexus 5X LGE bullhead
OS: Android N Preview 2 (build NPC91K)

`FATAL EXCEPTION: CameraTwoEngine
Process: com.commonsware.cwac.cam2.demo:cwac_cam2, PID: 24636
java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference
    at com.commonsware.cwac.cam2.CameraTwoEngine$RequestCaptureTransaction.capture(CameraTwoEngine.java:549)
    at com.commonsware.cwac.cam2.CameraTwoEngine$RequestCaptureTransaction.onCaptureProgressed(CameraTwoEngine.java:532)
    at java.lang.reflect.Method.invoke(Native Method)
    at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
    at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:148)
    at android.os.HandlerThread.run(HandlerThread.java:61)
`

commonsguy, 2016-04-15:
I do not have NDP2 on a 5X. I have it on a Nexus 6 and Nexus 9. I just tried taking a picture using the `demo-playground` app, and I did not encounter a crash.

Please provide the complete steps to reproduce the problem, so I can see if I can reproduce it on the hardware that I have.

simonnorberg, 2016-04-16:
To reproduce

1. Launch `demo` app
2. Press capture
3. Press green button to take picture

It crashes immediately after pressing the green button.

commonsguy, 2016-04-16:
Assuming that you are following the `demo` app instructions, you would be trying to take a picture with the rear-facing camera, in portrait mode.

If so, then I cannot reproduce the problem on:

- a Nexus 6 running NDP2
- a Nexus 9 running NDP2
- a Nexus 5X running Android 6

Hence, the problem would appear to be tied specifically to the 5X on NDP2.

Given that NDP2 is a developer preview, I am not going to race out and buy a spare 5X for the purposes of trying to reproduce this issue. I'll flag this as pending-hardware for now (where the "hardware" is N final getting on my 5X), so we will see where things stand at that point.

If you need to get past this problem in the short term, try `forceClassic()`. You are crashing in the `camera2` code; perhaps the classic `Camera` API will work better.

If I misunderstood your instructions for how to reproduce the error, please let me know, and I can try again.

simonnorberg, 2016-04-17:
Thanks for your help!

I also tried `demo-playground` with different settings and it crashes in the same way. Only `forceClassic()` works.

Let's see what happens in the next N release.

LetaiefAymen, 2016-06-09:
I had the same issue using a Nexus 5X with Android N Preview 2, the problem is the same:
the result of "result.get(CaptureResult.CONTROL_AF_STATE)" is always null and therefore can't be casted to int value. a workaround is to put a default value to "autoFocusState" variable and be sure that "result.get(CaptureResult.CONTROL_AF_STATE)" is not null when setting the value.
The only thing I don't know what is the best choice to put as a default value for the autoFocusState.

commonsguy, 2016-06-09:
@LetaiefAymen Once Android N ships in final form and I get it on a 5X, I will be happy to work on this.

op-and, 2016-07-11:
I am seeing the same NPE with Android N (bullhead npd56n) on a Nexus 5X. Wondering if you have any updates on the issue? Thanks.

commonsguy, 2016-07-12:
@op-and As I wrote previously, once Android N ships in final form and I get it on a 5X, I will be happy to work on this. This assumes that the problem still exists at that point.

op-and, 2016-07-12:
Thanks @commonsguy .

aznSlayer, 2016-08-30:
Hey @commonsguy any updates on this?

commonsguy, 2016-08-30:
@aznSlayer My Nexus 5X has not received the Android 7.0 update.

aznSlayer, 2016-08-30:
hmm weird all of mine did already, thanks for the prompt response!

oba2cat3, 2016-09-05:
Happened on a P6 device for one of my users today. @commonsguy did you get the update?
`Fatal Exception: java.lang.NullPointerException
Attempt to invoke virtual method 'int java.lang.Integer.intValue()' on a null object reference
 Raw Text
com.commonsware.cwac.cam2.CameraTwoEngine$RequestCaptureTransaction.capture (CameraTwoEngine.java:549)
com.commonsware.cwac.cam2.CameraTwoEngine$RequestCaptureTransaction.onCaptureProgressed (CameraTwoEngine.java:532)`

commonsguy, 2016-09-05:
@oba2cat3 I do not know what a "P6 device" is. Do you mean a Nexus 6P?

I cannot reproduce this problem on a Nexus 9 that has Android 7.0. I have not received updates for the Nexus 5X or 6P to Android 7.0.

alamode-travis-white, 2016-09-06:
I just started running into this issue as well after the latest update. Also on a Nexus 5X running Android 7.0 (NRD90M)

Your efforts are greatly appreciated @commonsguy

abkaplan07, 2016-09-08:
Using `forceClassic()` works for me in the interim. 

Using Nexus 5X, just updated to N this morning.

alecio, 2016-09-15:
The issue is consistently reproducible on Nexus 5X running Nougat.

For now we've placed the 'forceClassic()' alternative and it works, no side effects perceived.
Anyway, I will be looking forward for a official fix.

commonsguy, 2016-09-19:
Addressed in v0.6.7.

closed 2016-09-19 by commonsguy

alecio, 2016-09-20:
fix confirmed! Thanks @commonsguy