commonsguy/cwac-cam2 #185: Exception resetting focus [@Galaxy S7]

terao318, 2016-04-21:
Condition:
  app: demo
  occurrence: always, on all camera all orientation 
  device : Galaxy S7 samsungexynos8890 heeroltexx Android:6.0.1
  
com.commonsware.cwac.cam2.demo:cwac_cam2 W/CapturePictureTransaction: Exception resetting focus
                                                                                                     java.lang.IllegalStateException: Session has been closed; further changes are illegal.
                                                                                                         at android.hardware.camera2.impl.CameraCaptureSessionImpl.checkNotClosed(CameraCaptureSessionImpl.java:606)
                                                                                                         at android.hardware.camera2.impl.CameraCaptureSessionImpl.capture(CameraCaptureSessionImpl.java:170)
                                                                                                         at com.commonsware.cwac.cam2.CameraTwoEngine$CapturePictureTransaction.unlockFocus(CameraTwoEngine.java:678)
                                                                                                         at com.commonsware.cwac.cam2.CameraTwoEngine$CapturePictureTransaction.onCaptureCompleted(CameraTwoEngine.java:660)
                                                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                                                         at android.hardware.camera2.dispatch.InvokeDispatcher.dispatch(InvokeDispatcher.java:39)
                                                                                                         at android.hardware.camera2.dispatch.HandlerDispatcher$1.run(HandlerDispatcher.java:65)
                                                                                                         at android.os.Handler.handleCallback(Handler.java:739)
                                                                                                         at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                                         at android.os.Looper.loop(Looper.java:158)
                                                                                                         at android.os.HandlerThread.run(HandlerThread.java:61)

commonsguy, 2016-04-21:
> Galaxy S7 samsungexynos8890 heeroltexx Android:6.0.1

What is the `Build.PRODUCT` value for this device? `samsungexynos8890` appears to be a CPU name, and a search on Google turns up zero hits on `heeroltexx`. Alternatively, there is usually a model name listed in Samsung's edition of Settings > About.

terao318, 2016-04-21:
Hi,

This is info from Settings > About
Devise: Galaxy S7
Model : SM-G930FD
Android: 6.0.1

commonsguy, 2016-04-22:
Ah, OK, that wasn't even one of the possibilities that I had been considering. I will work on getting my hands on one of these. You might try `forceClassic()` and see if you have better luck, as your exception is coming from the `camera2` API.

terao318, 2016-04-22:
Thanks,
`forceClassic()` works flawlessly.
Do you have any info on hosts which `forceClassic()` may have issue?

By the way, I also encounter issue with Huawei mate 8, `requestPermissions()` does not work.
I'll make another issue ticket when I'll get the details.

commonsguy, 2016-04-22:
> Do you have any info on hosts which forceClassic() may have issue?

I am not aware of any that have problems with it.

commonsguy, 2016-05-01:
@terao318 I cannot reproduce your problem. The `demo` app works fine on the SM-G930FD, both for still photos and videos.

Please explain, **in detail**, how you are causing your crash.

commonsguy, 2016-05-13:
Closing as not reproducible.

closed 2016-05-13 by commonsguy