commonsguy/cwac-cam2 #187: Torch flash mode for Camera (including Video)

riodevista, 2016-04-24:
Hello!
Is it real to implement torch flash mode?
Thank you.

commonsguy, 2016-04-24:
Why would anyone want torch flash mode, in a library devoted to taking pictures and recording videos?

riodevista, 2016-04-24:
Thank you for the quick response.
I am developing an application in which I want to provide users torch flash mode while taking photos or recording videos. It is useful feature while using camera in the dark. Also stock camera on most devices does it.

commonsguy, 2016-04-24:
OK, I'll look into adding this in a future update.

vxhviet, 2016-05-05:
Dear @commonsguy, 
Thanks you for your concern on this feature. I would like to implement this feature but I'm kind of getting lost in the process. My initial thought is if I can somehow gain access to the underlying `Camera` of `ClassicCameraEngine` then I can do something like this:

```java
            params = camera.getParameters();
            params.setFlashMode(Parameters.FLASH_MODE_TORCH);
            camera.setParameters(params);
```

However both `ClassicCameraEngine` and `CameraTwoEngine` are singleton and there's no way to access the underlying `Camera` object.

Would you please give me a starting point on how to properly implement this?

Thanks you so much for your time.

vxhviet, 2016-05-05:
Hi @commonsguy, no need to worry about this anymore. I've partially solve this issue. All thanks to your awesome lib.

riodevista, 2016-05-05:
Hello @vxhviet!
Could you show the way you gain access to the `Camera` please?
Thank you!

vxhviet, 2016-05-05:
pls check out my updated comment

vxhviet, 2016-05-05:
Problem solved, I'll put complete code here when I've finished putting everything together.
In a meanwhile, look at @batrand comment in  issue #126 for the code snippet.

vxhviet, 2016-05-05:
Complete solution:

In `FlashMode` add the following:

```java
public enum FlashMode {
  ...
  TORCH(Camera.Parameters.FLASH_MODE_TORCH,
          CameraCharacteristics.FLASH_MODE_TORCH);
  ....

  static FlashMode lookupClassicMode(String classicMode) {
   ....
    else if (classicMode.equals(TORCH.getClassicMode())) {
      return(TORCH);
    }

    return(null);
  }

  static FlashMode lookupCameraTwoMode(int cameraTwoMode) {
    ....
    else if (cameraTwoMode==TORCH.getCameraTwoMode()) {
      return(TORCH);
    }

    return(null);
  }
  ....
}
```

In your fragment extended from `CameraFragment`:

```java
public View onCreateView(LayoutInflater inflater, ViewGroup container, Bundle savedInstanceState) {        
        ...
        if(mHasFlash){
            mFlashButton = (ImageView)v.findViewById(R.id.mycwac_cam2_flash_button);
            mFlashButton.setOnClickListener(new View.OnClickListener() {
                @Override
                public void onClick(View v) {
                    if(mIsFlashOn){
                        addFlashMode(FlashMode.OFF);
                        mIsFlashOn = false;
                    }else{
                        addFlashMode(FlashMode.TORCH);
                        mIsFlashOn = true;
                    }
                }
            });
        }
        ...
        if(mHasFlash){mFlashButton.setEnabled(false);}
        ...
}

private void addFlashMode(FlashMode flashmode){
        try {
            ctlr.stop();
            ArrayList<FlashMode> newFlashMode = new ArrayList<>();
            newFlashMode.add(flashmode); // This is the flash mode you want to change to
            ctlr.getEngine().setPreferredFlashModes(newFlashMode);
            ctlr.start();
        }
        catch (Exception e) {
            e.printStackTrace();
        }
}

public void onEventMainThread(CameraEngine.OpenedEvent event) {
        if (event.exception==null) {
            ...
            if(mHasFlash){mFlashButton.setEnabled(true);}
            ...
        }
        ...
}

private void recordVideo() {
        ...
        else {
            try {
                ...
                if(mHasFlash){
                    mFlashButton.setEnabled(false);
                    mFlashButton.setVisibility(View.INVISIBLE);
                }
                ...
            }
            ...
        }
}

private void setVideoFABToNormal() {
        if(mHasFlash){mFlashButton.setEnabled(true);}
        ...
}
```

Feel free to re-enable the visibility of your flash button. In my app, it return after finishing recording so it's unnecessary.

I've also try to enable/disable flash while recording but the `CameraController` get reset in `addFlashMode` so the recording get stopped whenever flash button is clicked. Because of that I kind of mimic the way the stock camera handle this torch feature. I haven't tested this carefully so feel free to try it and correct me if I'm wrong.

commonsguy, 2016-05-29:
Addressed in v0.6.0.

closed 2016-05-29 by commonsguy