commonsguy/cwac-cam2 #188: Why images are returned in zoomed out view

aftabsikander, 2016-04-26:
I have observed while opening camera, camera is already pre-configured to zoomed-in view, however if a picture is captured, it returns a zoomed-out view. However I am not using any zoom feature with intent which is as follows.

```
  cameraIntent = new CameraActivity.IntentBuilder(this)
                .skipConfirm()
                .focusMode(FocusMode.CONTINUOUS)
                .facing(Facing.BACK)
                .debug()
                .quality(AbstractCameraActivity.Quality.HIGH)
                .to(mCapturedImageUri)
                .debugSavePreviewFrame()
                .build();
```

PFA

- **camera_view_1** clearly shows camera screen before capturing the image which is already zoomed-in automatically.
[image]

- **picture_1** is the captured image, which was displayed in result of **camera_view_1**.
[image]

- **camera_view_2** also shows camera screen before capturing the image, in zoomed-in view.
[image]

- **picture_2** is the captured image, which was displayed in result of **camera_view_2**.
[image]

List of Devices which were used to produce this issue

**Device # 1**
```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```
**Device # 2**

```
Brand: samsung
DeviceModel: SM-G530H
Product: fortuna3gxx
Manufacturer: samsung
Android Version: 4.4.4
```

**Device # 3**

```
Brand: samsung
DeviceModel: SM-G530H
Product: fortuna3gxx
Manufacturer: samsung
Android Version: 5.0
```

**As per my understanding this issue is happening in all devices. What is the root cause for this weird behaviour?**

commonsguy, 2016-04-26:
The preview resolution and aspect ratio do not necessarily match the picture resolution and aspect ratio. What you are seeing is perfectly normal.

closed 2016-04-26 by commonsguy

aftabsikander, 2016-04-26:
Its still unclear to me why the camera view is zoomed in, In default camera application this never happens. See `camera_view_1 and picture_1` I am focusing water bottle and want that view to be captured only. To me this is not right behaviour of camera view. Is there a view for me to by pass this issue.

commonsguy, 2016-04-26:
> Its still unclear to me why the camera view is zoomed in

The aspect ratio of the preview image does not match the aspect ratio of the screen. Hence, part of the image will be cut off, in order to fill the screen with the rest of the preview image.

> Is there a view for me to by pass this issue.

You are welcome to use the camera APIs directly (e.g., `android.hardware.Camera`).

aftabsikander, 2016-04-27:
> The aspect ratio of the preview image does not match the aspect ratio of the screen. Hence, part of the image will be cut off, in order to fill the screen with the rest of the preview image.

This means this will be happening in every device. If I understand correctly when we open camera Activity you are not initializing camera view with device screen aspect ratio, hence this is occurring.

> You are welcome to use the camera APIs directly (e.g., android.hardware.Camera).

I believe If this issue is within this library then we need to handle this use case for many users of this library. instead of using camera API directly to by pass this issue.

commonsguy, 2016-04-27:
> you are not initializing camera view with device screen aspect ratio

The `TextureView` that is used for the camera `View` is set to fill the screen. However, the preview sizes supported by the camera do not have to match the size of the `TextureView`. It is up to the device manufacturer what preview sizes they choose to support, and we have no control over that. Similarly, we have no control over the size of the device screen (and, as a result, the size of the `TextureView`). These will rarely match.

aftabsikander, 2016-04-27:
Thanks for explaining, I understand and agree with your point. But all other application with uses camera API having `TextureView` this issue does not happen with those application.