commonsguy/cwac-cam2 #189: Images are getting displayed as landscape while taking in portrait mode

aftabsikander, 2016-04-27:
I am getting strange result when taking pictures in portrait mode, Picture was getting displayed in landscape. This is happening when we try to take picture while holding camera in the air (Portrait mode). I believe Camera view changes angles which are not respected while taking these type of pictures.`This same issue is happening in your demo and playground application as well`

[image]

List of Devices which were used to produce this issue

**Device # 1**
```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```
**Device # 2**

```
Brand: samsung
DeviceModel: SM-G530H
Product: fortuna3gxx
Manufacturer: samsung
Android Version: 4.4.4
```

**Device # 3**

```
Brand: samsung
DeviceModel: SM-G530H
Product: fortuna3gxx
Manufacturer: samsung
Android Version: 5.0
```

commonsguy, 2016-04-27:
Closing as duplicate of issue #15. If you feel that this is a separate issue, please explain in detail why and provide **the complete set of steps to reproduce the problem**.

closed 2016-04-27 by commonsguy

aftabsikander, 2016-04-27:
Its almost close to issue [#15](https://github.com/commonsguy/cwac-cam2/issues/15) I am getting `EXIF orientation` as `ExifInterface.ORIENTATION_UNDEFINED (0)`